realsenseai/realsense-ros #616: D435 realsense issues

brookdpg, 2019-02-14:
Hi Dorohni,

I have a D435 and I followed the instructions properly but the camera does not start. I did update to the latest possible firmware and it seems to identify it but is not able get any information from the sensors.

`ops@ops-XPS-15-9560:~/catkin_ws/src/realsense/realsense2_camera/launch$ roslaunch rs_camera.launch 
... logging to /home/ops/.ros/log/a3498386-3007-11e9-976b-9cb6d0dbfa9f/roslaunch-ops-XPS-15-9560-13846.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ops-XPS-15-9560:35115/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [13863]
process[camera/realsense2_camera-2]: started with pid [13864]
[ INFO] [1550115039.911743176]: Initializing nodelet with 8 worker threads.
[ INFO] [1550115039.955046462]: RealSense ROS v2.1.4
[ INFO] [1550115039.955086174]: Running with LibRealSense v2.18.1
[ INFO] [1550115040.082492848]: getParameters...
[ INFO] [1550115040.209137791]: setupDevice...
[ INFO] [1550115040.209193438]: JSON file is not provided
[ INFO] [1550115040.209233308]: ROS Node Namespace: camera
[ INFO] [1550115040.209265360]: Device Name: Intel RealSense D435
[ INFO] [1550115040.209294954]: Device Serial No: 819312072633
[ INFO] [1550115040.209322334]: Device FW version: 05.10.06.00
[ INFO] [1550115040.209344459]: Device Product ID: 0x0B07
[ INFO] [1550115040.209373938]: Enable PointCloud: Off
[ INFO] [1550115040.209419147]: Align Depth: Off
[ INFO] [1550115040.209460202]: Sync Mode: Off
[ INFO] [1550115040.209550923]: Device Sensors: 
[ INFO] [1550115040.209627893]: Stereo Module was found.
[ INFO] [1550115040.209682093]: RGB Camera was found.
[ INFO] [1550115040.209783094]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1550115040.209834350]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1550115040.209864700]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1550115040.210041435]: setupPublishers...
[ INFO] [1550115040.215599832]: Expected frequency for depth = 30.00000
[ INFO] [1550115040.289223714]: Expected frequency for infra1 = 30.00000
[ INFO] [1550115040.361502761]: Expected frequency for infra2 = 30.00000
[ INFO] [1550115040.414745494]: Expected frequency for color = 30.00000
[ INFO] [1550115040.454278087]: setupStreams...
 13/02 22:30:40,454 WARNING [140085558220672] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1550115040.455470608]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1550115040.455651801]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1550115040.455820395]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1550115040.457648928]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1550115040.462591045]: num_filters: 0
[ INFO] [1550115040.462967196]: RealSense Node Is Up!
[ INFO] [1550115040.462986230]: Setting Dynamic reconfig parameters.
[ INFO] [1550115040.632808227]: Done Setting Dynamic reconfig parameters.
`

doronhi, 2019-02-14:
Hi @brookdpg,
On first glance the log seems fine. Aren't there any warnings or errors later on?
What do you mean by "is not able get any information from the sensors."? Do you mean to say that you register to the topics and nothing is transmitted? 
what does `ros topic list` show?
`ros hz /camera/color/image_raw` shows nothing?

brookdpg, 2019-02-16:
No error later on.

brookdpg, 2019-02-16:
By "is not able to get any information from the sensors" I meant there is no streaming.

brookdpg, 2019-02-16:
as shown in the line
INFO] [1550115040.209783094]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1550115040.209834350]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1550115040.209864700]: (Accel, 0) sensor isn't supported by current device! -- Skipping...

brookdpg, 2019-02-16:
[
[image]
[image]

](url)

brookdpg, 2019-02-16:
After the last line where it says "Done setting Dynamic reconfig parameters" the process just hangs.

doronhi, 2019-02-17:
You are running D435 so the lines about the accel, gyro and fisheye are to be expected.
I’m sorry, I wasn’t clear. I wanted to know what is printed if you run : `rostopic list` and `rostopic hz` also if possible.

brookdpg, 2019-02-17:
ops@ops-XPS-15-9560:~$ rostopic list
/camera/RGB_Camera/parameter_descriptions
/camera/RGB_Camera/parameter_updates
/camera/Stereo_Module/parameter_descriptions
/camera/Stereo_Module/parameter_updates
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_color/image_raw/theora
/camera/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/theora/parameter_updates
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/aligned_depth_to_infra1/image_raw/compressed
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/compressedDepth
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/theora
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/theora/parameter_updates
/camera/aligned_depth_to_infra2/camera_info
/camera/aligned_depth_to_infra2/image_raw
/camera/aligned_depth_to_infra2/image_raw/compressed
/camera/aligned_depth_to_infra2/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_infra2/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_infra2/image_raw/compressedDepth
/camera/aligned_depth_to_infra2/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_infra2/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_infra2/image_raw/theora
/camera/aligned_depth_to_infra2/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_infra2/image_raw/theora/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/image_rect_raw/theora
/camera/infra2/image_rect_raw/theora/parameter_descriptions
/camera/infra2/image_rect_raw/theora/parameter_updates
/camera/realsense2_camera_manager/bond
/diagnostics
/rosout
/rosout_agg
/tf_static
ops@ops-XPS-15-9560:~$

brookdpg, 2019-02-17:
rostopic hz of which topic specifically do you require

doronhi, 2019-02-17:
These 2 topics:
/camera/color/camera_info
/camera/color/image_raw

brookdpg, 2019-02-17:
[image]

brookdpg, 2019-02-17:
seems normal to me...

doronhi, 2019-02-18:
It's a bit slow. I would expect around 30Hz, not 16 and that could be related to the firmware you use, but as I understand it, this is not the issue.
What is it that you do that makes you say "not able to get any information from the sensors"?

manhha1402, 2019-02-20:
The same problem to me with d435 camera: 
rostopic hz /camera/color/image_raw, rostopic hz /camera/aligned_depth_to_color/image_raw 
no new messages

manhha1402, 2019-02-20:
[image]

manhha1402, 2019-02-20:
It was a HW problem. It is fixed

brookdpg, 2019-02-22:
Figured out my problem. Thanks @dornhi

closed 2019-02-22 by brookdpg