realsenseai/realsense-ros #618: Benefits of using realsense rosbag instead of the regular rosbags

pedropgusmao, 2019-02-15:


doronhi, 2019-02-18:
The bagfile saved by realsense-viewer contains several topics.
The 'metadata' topic contains system time.
In python you could see it like that:
```
>>> import rosbag
>>> bag = rosbag.Bag(filename)
>>> mmsgs = [msg for topic, msg, t in bag.read_messages(topics=['/device_0/sensor_1/Color_0/image/metadata'])]
>>> systime = [msg for msg in mmsgs if 'system_time'==msg.key]
 
>>> import datetime
>>> datetime.datetime.fromtimestamp(float(systime[0].value)/1000)
datetime.datetime(2019, 2, 18, 0, 55, 19, 132249)
```
The timestamp you saw is most likely the device hardware time. i.e., the time since you plugged it in.

The benefit of using the realsense bag file is that you can run the realsense2_wrapper with it using the option `rosbag_filename`. The wrapper will publish the images, giving them the current ROS time and you could enable or disable different filters every run to try out the post-processing benefits.
If you use the bag file saved using **rosbag record -a** you are confined to the output of the wrapper, not to the output of the camera.

Regarding loss of frames, that really depends on the kind of machine you use. I would use a very powerful machine for the recordings, regardless of the machine you will use later on.

pedropgusmao, 2019-02-20:
Thanks, for this! 
So the hardware time will be when I plugged the camera in but not the the time when the first frame was recorded, right? Is there any way to recover this information? 
Again, 

Thanks!

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

pedropgusmao, 2019-03-24:
Sorry, I found the metadata you mentioned. Thanks!

closed 2019-03-24 by pedropgusmao

anug7, 2019-07-03:
@pedropgusmao , Did you find the meta-information needed to do the post processing like alignment?