realsenseai/realsense-ros #620: Cannot change more than one dynamic reconfigure parameter at the same time

stwirth, 2019-02-15:
The current implementation (ab)uses the `level` of a parameter as index of the `rs2_option`.
This is problematic as the dynamic reconfigure callback assumes that the provided `level` only contains the index of _one_ parameter. In fact, `set_option` is only called once in the reconfigure callback, instead of for every changed parameter in the DDMap:

https://github.com/intel-ros/realsense/blob/a036d81bcc6890658104a8de1cba24538effd6e3/realsense2_camera/src/base_realsense_node.cpp#L299-L303
This works well when you just alter one parameter at a time in `rqt_reconfigure` but fails when you use dynamic reconfigure programmatically and make a `set_parameters` service call that contains more than one change.

doronhi, 2019-02-18:
I managed to reproduce your issue using:
```
rosrun dynamic_reconfigure dynparam set /camera/RGB_Camera "{'Brightness':3, 'Contrast':63}"
```
I see what you mean. Do you propose a way to change it?
Suppose we use bits in **level** as flags. dynamic_reconfigure defines the callback with **u_int32 level**.
librealsense already defines 47 optional options (rs_options.h)

stwirth, 2019-02-18:
@doronhi I'm glad you were able to reproduce this.
The change I'd propose is to not use the `level` at all but use the parameter names from the incoming `DDMap` directly:

```C++
void BaseRealSenseNode::callback(const ddynamic_reconfigure::DDMap& map, int level, rs2::options sensor) {
    ROS_INFO_STREAM("Received reconfigure callback");
    for (auto i = 0; i < RS2_OPTION_COUNT; i++)
    {
        rs2_option option = static_cast<rs2_option>(i);
        std::string option_name = create_graph_resource_name(rs2_option_to_string(option));
        DDPtr param = at(map, option_name.c_str());
        if (param) {
            double value = param->getValue().toDouble();
            ROS_INFO_STREAM("setting option '" << option_name << "' to value " << value);
            sensor.set_option(option, value);
        }
    }
}
```

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@stwirth Can https://github.com/IntelRealSense/realsense-ros/pull/693 work for you? Thanks!

stwirth, 2019-06-17:
I have not tested it but the code change looks like it fixes the issue. Thanks!

closed 2019-06-17 by stwirth