realsenseai/realsense-ros #621: error compiling ddynamic reconfigure.

FPSychotic, 2019-02-17:
Im getting this error with the last two versions of ros wrapper and SDK. Im using Aeeon UP with 4.15 kernel,and SDK with realsense driver correctly installed. I have openVINO installed too. 

---------------------------------------------------------------------------------------------------------------------------

Scanning dependencies of target roscpp_generate_messages_lisp
[  8%] Built target roscpp_generate_messages_lisp
Scanning dependencies of target roscpp_generate_messages_py
[  8%] Built target roscpp_generate_messages_py
Scanning dependencies of target _realsense2_camera_generate_messages_check_deps_Extrinsics
[  8%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
Scanning dependencies of target _realsense2_camera_generate_messages_check_deps_IMUInfo
[  8%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
Scanning dependencies of target sensor_msgs_generate_messages_cpp
[  8%] Built target sensor_msgs_generate_messages_cpp
[ 11%] Building CXX object realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure.dir/src/param/dd_double_param.cpp.o
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/genjava/cmake/../../../lib/genjava/genjava_gradle_project.py", line 14, in <module>
    genjava.main(sys.argv)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genjava/genjava_main.py", line 82, in main
    gradle_project.create(args.package, args.output_dir)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genjava/gradle_project.py", line 152, in create
    raise IOError("could not find %s among message packages. Does the package have a <build_depend> on message_generation in its package.xml?" % msg_pkg_name)
IOError: could not find ddynamic_reconfigure among message packages. Does the package have a 
<build_depend> on message_generation in its package.xml?
realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure_generate_messages_java_gradle.dir/build.make:63: recipe for target 'realsense-2.2.0/ddynamic_reconfigure/java/ddynamic_reconfigure/build.gradle' failed
make[2]: *** [realsense-2.2.0/ddynamic_reconfigure/java/ddynamic_reconfigure/build.gradle] Error 1
CMakeFiles/Makefile2:341: recipe for target 'realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure_generate_messages_java_gradle.dir/all' failed
make[1]: *** [realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure_generate_messages_java_gradle.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 13%] Building CXX object realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure.dir/src/param/dd_bool_param.cpp.o
[ 16%] Building CXX object realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure.dir/src/param/dd_string_param.cpp.o
[ 19%] Building CXX object realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure.dir/src/param/dd_enum_param.cpp.o
[ 22%] Building CXX object realsense-2.2.0/ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure.dir/src/dd_param.cpp.o
[ 25%] Linking CXX shared library /home/ihexa/realsense2_ws/devel/lib/libddynamic_reconfigure.so
[ 25%] Built target ddynamic_reconfigure
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Drkstr, 2019-02-20:
Any updates on this? I've got the same issue on my laptop (Ubuntu 16.04).

stwirth, 2019-02-20:
Maybe https://github.com/intel-ros/realsense/pull/624 fixes this?

Drkstr, 2019-02-21:
@stwirth that worked. Thankyou

closed 2019-02-24 by doronhi