realsenseai/realsense-ros #626: d435i accel and gyro have really diffrent time stamp

Dunkem, 2019-02-20:
when I use the d435i for vio application, but the time stamp is really untrusted between accel and gyro.
the gyro time stamp is below:
header: 
  seq: 425
  stamp: 
    secs: 3101275587
    nsecs: 883002758
  frame_id: "camera_gyro_frame"
while the accel tiem stamp is:
header: 
  seq: 424
  stamp: 
    secs: 1550637828
    nsecs: 963475704
  frame_id: "camera_accel_frame"
could anyone tell me why cause this huge time differences between accel and gyro

Shashika007, 2019-02-20:
@Dunkem  They are totally independent and also not synchronized. I'm testing D435i for VIO also. I saw the discontinuity after select the unite_imu_method as linear_interpolation (https://github.com/intel-ros/realsense/issues/577). I think the topics are not published with the same timestamp. I'm using the latest firmware and ros wrapper. But, still they have not fixed the problem i guess.

Dunkem, 2019-02-20:
@Shashika007 I know this two sensor is not synchronized. but actually the time stamp's secs should be the same or just a little difference, but the gyro time stamp's secs seems twice accel time stamp's secs.

RigerLee, 2019-02-20:
In not sure if setting "initial_reset" in your launch file to true works.
But I've ever encountered such problem between IMU and color image and solved it this way.

Shashika007, 2019-02-21:
@RigerLee  I already set the initial_reset, I did not noticed any difference. 
@Dunkem have you noticed this ? the gyro and accel do not publish the topics continuously with same time delay. Like there are some big gaps between two readings. 
[image]
[image]
@RigerLee @Dunkem Do you guys have any idea about this issue ?

Dunkem, 2019-02-22:
@RigerLee I also already set the initial_reset=true,but it seems not work.
@Shashika007 I don't know why cause this big gaps.And can you tell me how do you get this image.

Shashika007, 2019-02-22:
@Dunkem you can record a bag file that includes imu data and check it using rqt.

doronhi, 2019-02-24:
Hi @Dunkem 
The gyro and accel topic are not synchronized but the both have the same clock therefor their timestamps should be very close.
Your gyro timestamp seems almost exactly **twice** the accel timestamp.
What firmware do you have? librealsense version? realsense2_wrapper version?
What machine do you run it on?
Can you reproduce this phenomena?

Dunkem, 2019-02-26:
@doronhi Thank you for your reply！My tun it in my laptop Latitude 7380，the librealsense version is 2.18.1，and I clone the ros wrapp from https://github.com/intel-ros/realsense just 6 days ago.I don’t know what the firmware version is.

Dunkem, 2019-02-26:
@doronhi and  the gyro timestamp is twice of the accel timestamp whenever I run it

Shashika007, 2019-02-26:
@Dunkem you can check your firmware version by intel-realsense-dfu –p (if you don't have dfu install by sudo apt-get install intel-realsense-dfu*). Also, you can update new firmware by following below mentioned guide,
 https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

Dunkem, 2019-02-26:
@Shashika007 Very thanks for your guiding,and my firmware version is 5.10.13.0 @doronhi

doronhi, 2019-02-26:
@Dunkem 
Can you copy here the command you use to start the realsense2_camera node? what launch file and additional parameters?
What command do you type to print the results you get?

Dunkem, 2019-02-27:
@doronhi I run it with command “roslaunch realsense2_camera rs_camera.launch” and I set the parameter   "enable_sync=true","unite_imu_method=copy"

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

Jwenda, 2019-03-29:
Hi,
I have the same problem as @Dunkem has. Have you already fixed the issue, or is it still pending?

Thank you,
Jan

Dunkem, 2019-03-29:
@Jwenda I sorry but this problem still exist in my device but I replace my computer with jetson tx2 this problem disappear, I don't know why. And because the gap between two readings vio doesn't work well.

Jwenda, 2019-03-29:
@Dunkem Hmm.. interesting. We were running on Intel NUC which doesn't work, but the normal PC works. I will have to do some modification because I have to run on Intel NUC due to our hardware. If anybody finds the reason, please let me know.

Thanks,
Jan

Jwenda, 2019-04-18:
@Dunkem I managed to get rid of different timestamp by compiling the librealsense by myself, but it was on Lenovo PC with dual boot. But now I face another issue, with cycled timestamp, I will have to create a new issue.

doronhi, 2019-04-18:
@Jwenda , check out [issue 703](https://github.com/intel-ros/realsense/issues/703) about repeating time stamps.

Jwenda, 2019-04-18:
@doronhi Yes, thank you. I have already found it while creating a new issue. Glad it has been solved.

Dunkem, 2019-04-19:
it also works on my computer， Thank you @Jwenda

closed 2019-04-19 by Dunkem

Jwenda, 2019-06-05:
I have to correct my answer. I also had to run my Ubuntu on lower kernel version(4.15.0.47) instead of (4.18.0.20) to have a correct timestamp. I know that Intel Realsense SDK does not support the higher kernel than 4.16., so it might be the issue also.