realsenseai/realsense-ros #628: Add guards around clang-specific pragmas

jarvisschultz, 2019-02-20:
There are a few clang-specific pragmas used throughout the code that throw warnings when compiling with GCC. For example, see [the beginning fo dd_enum_params.cpp](https://github.com/intel-ros/realsense/blob/3e25326b4128d37b15d1e11085dd2f19d3cf1a66/ddynamic_reconfigure/src/param/dd_enum_param.cpp#L1-L5). Should all of these be wrapped with appropriate `#ifdef` calls? E.g.

```cpp
#ifdef __clang__
#pragma clang diagnostic push
#pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
#pragma clang diagnostic ignored "-Wunknown-pragmas"
#pragma ide diagnostic ignored "modernize-loop-convert"
#pragma ide diagnostic ignored "modernize-use-auto"
#endif
```
suppresses the error.

doronhi, 2019-02-21:
Could be a nice idea. The warnings do bother me but I never got around to handle it yet. Any help will be appreciated.

jarvisschultz, 2019-02-21:
Added a PR. Let me know if there are any issues.

commit 567b5ba5 (2019-03-04): Fix #628 - added guards around clang-specific pragmas (#630)

Also added a guard around an OpenMP pragma
closed 2019-03-04 by doronhi