realsenseai/realsense-ros #634: Pointcloud not synced with RGB Image

piraka9011, 2019-02-25:
Hi, I've been trying to work with the D435 but the documentation on working with transforms and TF in general is sketchy in my opinion. Best I found was [this](https://realsense.intel.com/how-to-multiple-camera-setup-with-ros/) and even then, the script used is buggy.

I get a TF mismatch between the pointcloud's `camera_depth_optical_frame` and the actual image by 90 degrees. I am using apriltags2_ros to get the object. No params in the realsense launch file was changed (using `rs_rgbd.launch`). I am publishing my camera_link with respect to the robot base. Is this an issue with the source code or should I just republish the depth TF?

Image below shows how object is in one frame, but pointclouds displayed somewhere else.

[image]

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

piraka9011, 2019-03-07:
A potential workaround for this is just to republish the frames using the `tf_ros` `static_transform_publisher`. However, I just wanted to confirm whether this is normal behavior and whether a interface exists to setup the camera transforms for a robot before closing.

doronhi, 2019-03-10:
Hi @piraka9011 ,
I am so sorry. I tried but could not understand what the issue was nor what is it that I see in the picture. Could you publish the workaround you found and maybe it will clear the issue up for me?
I wish do understand it before you close the issue, even if you managed to come up with a workaround solution.
Thanks,

piraka9011, 2019-03-22:
Sure, I think this was more of a confusion from my end. Based on the documentation, I thought the frame I had to update for the camera was `camera_link`, however, the frame I had to update was `camera_depth_optical_frame`. Working with the `camera_link` frame causes the point cloud to be rotated 90 degrees in the y-axis w.r.t the world. 
I updating the frames using a ROS [static transform publisher.](https://wiki.ros.org/tf#static_transform_publisher).

closed 2019-03-22 by piraka9011