realsenseai/realsense-ros #635: No RGB and IMU data on D435i

abylikhsanov, 2019-02-26:
I am running D435i on Odroid Xu4 and I have installed librealsense v2.18.1 and SDK 2.2.0. I am runing Ubuntu Mate 18.04. 
When I try to run the ROS wraper I cannot get RGB data (regular camera) and IMU:

```
[ INFO] [1551194562.321732022]: Initializing nodelet with 8 worker threads.
[ INFO] [1551194563.319228071]: RealSense ROS v2.2.0
[ INFO] [1551194563.319358910]: Running with LibRealSense v2.18.1
[ INFO] [1551194563.627934726]: getParameters...
[ INFO] [1551194564.538925204]: setupDevice...
[ INFO] [1551194564.539115336]: JSON file is not provided
[ INFO] [1551194564.539309635]: ROS Node Namespace: camera
[ INFO] [1551194564.539554852]: Device Name: Intel RealSense USB2
[ INFO] [1551194564.539709400]: Device Serial No: 843212070170
[ INFO] [1551194564.539886031]: Device FW version: 05.10.13.00
[ INFO] [1551194564.540034245]: Device Product ID: 0x0AD6
[ INFO] [1551194564.540214710]: Enable PointCloud: On
[ INFO] [1551194564.540379133]: Align Depth: Off
[ INFO] [1551194564.540538889]: Sync Mode: On
[ INFO] [1551194564.542241036]: Device Sensors: 
[ INFO] [1551194564.543124736]: Stereo Module was found.
[ INFO] [1551194564.543389912]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.543540584]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.543676298]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.543800803]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.543941599]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.544072104]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.544237277]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1551194564.545057891]: setupPublishers...
[ INFO] [1551194564.565615325]: Expected frequency for depth = 30.00000
[ INFO] [1551194564.606545145]: Expected frequency for infra1 = 30.00000
[ INFO] [1551194564.622868880]: Expected frequency for infra2 = 30.00000
[ INFO] [1551194564.651048307]: setupStreams...
 26/02 15:22:44,656 WARNING [3069612192] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1551194564.667300289]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1551194564.668452124]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ WARN] [1551194564.668735259]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 640, Height: 480, FPS: 30
[ INFO] [1551194564.674493724]: insert Depth to Stereo Module
[ INFO] [1551194564.675289004]: insert Infrared to Stereo Module
[ INFO] [1551194564.716073777]: Add Filter: pointcloud
[ INFO] [1551194564.740642319]: num_filters: 1
[ INFO] [1551194564.741586855]: SELECTED BASE:Depth, 0
[ INFO] [1551194564.744876353]: RealSense Node Is Up!
[ INFO] [1551194564.745029483]: Setting Dynamic reconfig parameters.
 26/02 15:22:44,867 WARNING [2726437904] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ INFO] [1551194564.872984511]: Done Setting Dynamic reconfig parameters.
[ WARN] [1551194564.878694266]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

```

Shashika007, 2019-02-27:
@abylikhsanov is Odroid Xu4 has usb 3.0 ? can you launch realsense-viewer and display the IMU and visual data ?

abylikhsanov, 2019-02-27:
Yes it has, I can display everything except simple RGB

doronhi, 2019-02-27:
According to the log you published, the device was connected as a USB 2 and it is behavior in this case is not well defined.
Assuming you connect it to a USB 3 jack, you should try to reconnect again and check how it was connected using linux tools, like this:
```
lsusb -d 8086: -v | grep -i bcdUSB
```
If it shows:
`bcdUSB               2.10` That's not good.
`bcdUSB               3.20` is good.

If you can run realsense-viewer, it also shows if it was connected as a usb2.1 or 3.2

abylikhsanov, 2019-02-27:
@doronhi You were right about the USB, I have used a different USB port and now the device is being recognized as USB 3.20

However, when I try to launch rs_launch I get an error that the device is not found.

Here is the output of dmesg when I plug in the camera:

```
[   41.820044] usb 4-1.2: new SuperSpeed USB device number 3 using xhci-hcd
[   41.840968] usb 4-1.2: New USB device found, idVendor=8086, idProduct=0b3a
[   41.840979] usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   41.840987] usb 4-1.2: Product: Intel(R) RealSense(TM) Depth Camera 435i
[   41.840994] usb 4-1.2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[   41.841001] usb 4-1.2: SerialNumber: 852323053439
[   41.906634] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[   41.906726] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[   41.906948] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[   41.912374] uvcvideo: Unable to create debugfs 4-3 directory.
[   41.912780] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[   41.912791] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[   41.912799] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[   41.912807] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[   41.913211] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input1
[   41.914053] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[   41.914066] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[   41.914487] uvcvideo: Unable to create debugfs 4-3 directory.
[   41.914687] uvcvideo 4-1.2:1.3: Entity type for entity Processing 7 was not initialized!
[   41.914698] uvcvideo 4-1.2:1.3: Entity type for entity Extension 8 was not initialized!
[   41.914707] uvcvideo 4-1.2:1.3: Entity type for entity Camera 6 was not initialized!
[   41.914978] usbcore: registered new interface driver uvcvideo
[   41.914984] USB Video Class driver (1.1.1)
[   41.918937] usbcore: registered new interface driver usbhid
[   41.918945] usbhid: USB HID core driver

```

And the roslog tells this:

```
[ERROR] [1551261069.156594566]: No RealSense devices were found!
[ERROR] [1551261069.156777548]: Terminating RealSense Node...
[FATAL] [1551261069.373191382]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'

```

doronhi, 2019-02-27:
It seems there is a problem with your librealsense-dkms installation.
Checkout the installation instructions in: https://github.com/doronhi/librealsense/blob/master/doc/distribution_linux.md

You may need to reboot afterwards.

abylikhsanov, 2019-02-28:
I tried to install it but seems they do not support arm:
```
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' doesn't support architecture 'armhf'
```

abylikhsanov, 2019-02-28:
@doronhi But isn't building from source installs that? I just did:
https://github.com/doronhi/librealsense/blob/master/doc/installation.md

And I also tried to run both scripts (odroid and odroid-4.14) but in both cases the same results.

abylikhsanov, 2019-03-01:
So nothing I can do? Shall we say it does not work on Odroid with the current patch?

abylikhsanov, 2019-03-04:
I have fixed the error. The problem was with the Kernel patch and I simply fixed it a little bit for D435i to work on Odroid XU4.

closed 2019-03-04 by abylikhsanov

jsseng, 2019-05-17:
Could you describe the fix?  I am running into this same problem.

florianspy, 2019-11-12:
I recently had the same issue with an wrong usb cable even thought it was supposed to be usb3.0 switching to the version shipped with the device sloved the issue