realsenseai/realsense-ros #640: SLAM Demo

jattiogbe, 2019-03-01:
Hi everyrone,

Has anyone managed to perform efficient SLAM with D435 / D435I in a room or outside ?
Would you please share a video or some pictures of the results ?

Thanks in advance.

diegomrt, 2019-03-01:
Sure, you can have a look at [this video](https://urjc-my.sharepoint.com/:v:/g/personal/diego_martin_martin_urjc_es/ESJYuJGC9TxBnZyRbxqbuL8B3gtV43JnPgHeYEtDeHPSrA?e=pg5chi). It is ROS-based RTAB-MAP using a D435.

jattiogbe, 2019-03-01:
Nice video, thanks @diegomrt !

closed 2019-03-04 by doronhi

Huntrr, 2019-03-28:
@diegomrt would you be willing to share the D435 bag that generated this and the configuration you used? Your results are way better than mine and I'm trying to figure out what the difference might be.