realsenseai/realsense-ros #642: Docker: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

juergensturm, 2019-03-01:
I want to run the realsense node in a Docker container, but I keep getting the following error:

```
[ INFO] [1551446004.679784656]: RealSense Node Is Up!
[ INFO] [1551446004.679832510]: Setting Dynamic reconfig parameters.
 01/03 13:13:24,683 ERROR [139674062460800] (types.h:214) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
[ERROR] [1551446004.683362375]: An exception has been thrown: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
[FATAL] [1551446004.739934165]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 157, exit code -11, cmd /opt/ros/lunar/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/root/.ros/log/c920a6c0-3c23-11e9-9dd1-0242c0a80902/camera-realsense2_camera_manager-2.log].
```

My Dockerfile looks as follows:

```
FROM library/ros:lunar-perception-xenial

ENV DEBIAN_FRONTEND noninteractive

# Install additional dependencies
RUN \
  apt-get update && \
  apt-get -y install gnupg2 git nano software-properties-common \
    ros-lunar-rgbd-launch ros-lunar-diagnostic-updater && \
  rm -rf /var/lib/apt/lists/*

# Install realsense
RUN \
  apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE && \
  add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u && \
  apt-get update && \
  apt-get -y install librealsense2-utils librealsense2-dev && \
  rm -rf /var/lib/apt/lists/*

# Build and install ROS wrapper for Realsense
RUN \
  mkdir -p /catkin_ws/src && \
  cd /catkin_ws/src/ && \
  git clone -b 2.1.4 https://github.com/intel-ros/realsense.git && \
  bash -c "source /opt/ros/lunar/setup.bash && catkin_init_workspace" && \
  cd /catkin_ws && \
  bash -c "source /opt/ros/lunar/setup.bash && catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release install" && \
  rm -rf /catkin_ws/build
```

Build and run using:
```
$ docker build . -t realsense
$ docker run --rm -ti --name realsense --privileged -v /dev:/dev realsense bash -c "source catkin_ws/devel/setup.bash && roslaunch realsense2_camera rs_camera.launch"
```

It seems that I can circumvent this issue by adding a few sleep statements in base_realsense_node.cpp as follows (it still occasionally crashes in other places, but by far less frequently):
```
--- a/realsense2_camera/src/base_realsense_node.cpp
+++ b/realsense2_camera/src/base_realsense_node.cpp
@@ -4,6 +4,7 @@
 #include <algorithm>
 #include <mutex>
 #include <tf/transform_broadcaster.h>
+#include <thread>
 
 using namespace realsense2_camera;
 using namespace ddynamic_reconfigure;
@@ -14,6 +15,8 @@ using namespace ddynamic_reconfigure;
 #define OPTICAL_FRAME_ID(sip) (static_cast<std::ostringstream&&>(std::ostringstream() << "camera_" << STREAM_NAME(sip) << "_optical_frame")).str()
 #define ALIGNED_DEPTH_TO_FRAME_ID(sip) (static_cast<std::ostringstream&&>(std::ostringstream() << "camera_aligned_depth_to_" << STREAM_NAME(sip) << "_frame")).str()
 
+// NOTE(jsturm): We introduce a small delay before every query to the UVC device.
+constexpr int kDelayMs = 100;
 SyncedImuPublisher::SyncedImuPublisher(ros::Publisher imu_publisher, std::size_t waiting_list_size):
             _publisher(imu_publisher),
             _waiting_list_size(waiting_list_size)
@@ -177,6 +180,7 @@ void BaseRealSenseNode::publishTopics()
 
 bool is_checkbox(rs2::options sensor, rs2_option option)
 {
+    std::this_thread::sleep_for (std::chrono::milliseconds(kDelayMs));
     rs2::option_range op_range = sensor.get_option_range(option);
     return op_range.max == 1.0f &&
         op_range.min == 0.0f &&
@@ -185,6 +189,7 @@ bool is_checkbox(rs2::options sensor, rs2_option option)
 
 bool is_enum_option(rs2::options sensor, rs2_option option)
 {
+    std::this_thread::sleep_for (std::chrono::milliseconds(kDelayMs));
     rs2::option_range op_range = sensor.get_option_range(option);
     if (op_range.step < 0.001f) return false;
     for (auto i = op_range.min; i <= op_range.max; i += op_range.step)
@@ -197,6 +202,7 @@ bool is_enum_option(rs2::options sensor, rs2_option option)
 
 bool is_int_option(rs2::options sensor, rs2_option option)
 {
+    std::this_thread::sleep_for (std::chrono::milliseconds(kDelayMs));
     rs2::option_range op_range = sensor.get_option_range(option);
     return (op_range.step == 1.0);
 }
```

Furthermore, I noticed a second, less severe problem. When the ROS node is configured to talk to a ROS_MASTER_URI outside of the current container, the camera nodelet manager crashes with a segfault. I can circument this by changing the launch file to use standalone mode instead:

```
diff --git a/realsense2_camera/launch/includes/nodelet.launch.xml b/realsense2_camera/launch/includes/nodelet.launch.xml
index 1839024..2de5b39 100644
--- a/realsense2_camera/launch/includes/nodelet.launch.xml
+++ b/realsense2_camera/launch/includes/nodelet.launch.xml
@@ -78,8 +78,7 @@
   <arg name="unite_imu_method"         default="none"/> <!-- Options are: [none, copy, linear_interpolation] -->
 
 
-  <node pkg="nodelet" type="nodelet" name="$(arg manager)" args="manager" output="screen"/>
-  <node pkg="nodelet" type="nodelet" name="realsense2_camera" args="load realsense2_camera/RealSenseNodeFactory $(arg manager)">
+  <node pkg="nodelet" type="nodelet" name="realsense2_camera" args="standalone realsense2_camera/RealSenseNodeFactory" output="screen" required="true">
     <param name="serial_no"                type="str"  value="$(arg serial_no)"/>
     <param name="json_file_path"           type="str"  value="$(arg json_file_path)"/>
     <param name="rosbag_filename"          type="str"  value="$(arg rosbag_filename)"/>
diff --git a/realsense2_camera/src/base_realsense_node.cpp b/realsense2_camera/src/base_realsense_node.cpp
index 10ab8bf..ab88267 100644
```

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

juergensturm, 2019-03-14:
Just realized that this seems to be a duplicate of https://github.com/IntelRealSense/librealsense/issues/2850.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@juergensturm Did you get it through as the proposal in https://github.com/IntelRealSense/librealsense/issues/2850? Could you please update? Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@juergensturm Any other questions about this ticket?

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@juergensturm  Didn't get response from you for a few weeks. Will close the ticket if no other questions.

closed 2019-06-03 by RealSenseCustomerSupport