realsenseai/realsense-ros #648: [FATAL] Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'

XXLbot, 2019-03-04:
Hello 

**Required informations**
Operating System & Version		Ubuntu 16.04.6 LTS
Kernel Version (Linux Only)          	4.4.38-v7+
ROS Version				        Kinetic 1.12.14
Platform                                		Raspberry Pi 3B
Camera Model			                D435
Firmware Version                  		5.11.1.0
RealSense SDK Version                     2.17.1
RealSense ROS Wrapper Version	2.1.3 installed  (2.1.2 displayed in log)
Language                           		(C++ and Python)

**RealSense ROS Wrapper  2.1.3 installation**
catkin build realsense2_camera			
  [build] Found '2' packages in 0.0 seconds.
  Finished  <<< ddynamic_reconfigure 
  Finished  <<< realsense2_camera
  1st   time I get some warnings
  2nd  time I get ''[build] Summary: All 2 packages succeeded!''

sudo apt-get install ros-kinetic-rgbd-launch

**Problem description**
**Test #1**
roslaunch  realsense2_camera  rs_camera.launch
Log attached 
[RealSense ROS Wrapper Test#1 - log v1 2019.02.28.txt](https://github.com/intel-ros/realsense/files/2926134/RealSense.ROS.Wrapper.Test.1.-.log.v1.2019.02.28.txt)

[ INFO] [.]: RealSense ROS v2.1.2
[ INFO] [.]: Running with LibRealSense v2.17.1
[ INFO] [.]: Device Name: Intel RealSense D435
[ INFO] [.]: Device FW version: 05.11.01.00
[ INFO] [.]: Enable PointCloud: Off
[ INFO] [.]: Stereo Module was found.
[ INFO] [.]: RGB Camera was found.
[ INFO] [.]: setupStreams...
[ INFO] [.]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [.]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [.]: RealSense Node Is Up!
[ INFO] [.]: Setting Dynamic reconfig parameters.
[ INFO] [.]: Done Setting Dynamic reconfig parameters.
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
It's a librealsense message . I get this message as described in Issue #3327

[ WARN] [.]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 640, Height: 480, FPS: 30
WARNING [] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [.]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
[ERROR] [.]: Hardware Notification:USB SCP overflow,1.54947e+12,Error,Hardware Error
RealSense SDK installed from source
Unfortunatelly uvcvideo and videodev not loaded 
This is the reason why I opened Issue #3327... 

Any clarification about WARNING & ERROR messages ?

**Test #2**
I try to start PointCloud feature and get the associated topic
roslaunch realsense2_camera rs_camera.launch   enable_pointcloud:=true
Log attached   
[RealSense ROS Wrapper Test#2 - log v1 2019.03.01.txt](https://github.com/intel-ros/realsense/files/2926137/RealSense.ROS.Wrapper.Test.2.-.log.v1.2019.03.01.txt)

[ INFO] [.]: Enable PointCloud: On
[ INFO] [.]: Add Filter: pointcloud
[ INFO] [.]: num_filters: 1
I understand that PointCloud can be started with success … Correct ?

**Test #3**
I try to start PointCloud feature/topic playing with paremters in order to reduce FPS (too high FPS should not be sustainable by a small Raspberry Pi 3B!)
roslaunch realsense2_camera rs_camera.launch   enable_pointcloud:=true  depth_fps:=10 color_fps:=5 infra1_fps:=5 infra2_fps:=5
Log attached   
[RealSense ROS Wrapper Test#3 - log v1 2019.03.01.txt](https://github.com/intel-ros/realsense/files/2926138/RealSense.ROS.Wrapper.Test.3.-.log.v1.2019.03.01.txt)

I get the following error
[FATAL] [.]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died 
Where comes this fatal error from ?!

Thanks in advance for your help
Regards

doronhi, 2019-03-04:
Hi @XXLbot 
1. The first issue seems like the one we see when the device is recognized as USB2 although it is really connected to USB3. You can confirm using `lsusb` or by running `realsense-viewer`. Usually disconnecting and reconnecting the device fix the problem.
Also make sure you installed the dkms correctly.
2. That is correct. 
3. The device can't send images in any frame rate. To see available profiles for each sensor type `rs-enumerate-devices` In any case, the point cloud will be available by the lower frame rate so there is no point in setting them to different frequencies.
The crash probably happen because the depth stream was not available. Starting with release 2.2.0, it is not mandatory, although you will need it for pointcloud.

XXLbot, 2019-03-05:
@ doronhi  Thanks for your update
1. As I mentioned the platform is Raspberry Pi 3B, so based on ARM, so librealsense2 DKMS is not available from Intel. 
2. I ran
roslaunch realsense2_camera rs_camera.launch  enable_color:=false enable_infra1:=false  enable_infra2:=false enable_imu:=false enable_fisheye:=false  enable_depth:=true depth_width:=480  depth_height:=270  depth_fps:=6 enable_pointcloud:=true
and I still get the following messages
From librealsense2 
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
WARNING [] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
From  realsense2_camera 
[ WARN] [.]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
Any suggestion ?
Regards

XXLbot, 2019-03-05:
Last but not least, I spent hours looking at GitHub issues and found  #2029  #1253 #1760 #433 #409 
#432  #528 .... so it's a recurrent problem that the users are facing....
Regards

XXLbot, 2019-03-06:
I followed comment in librealsense2 issue #1132
and choosed to build the SDK from source with CMake flag -DFORCE_LIBUVC=true. 

Old test (without -DFORCE_LIBUVC=true), I got 
realsense-viewer
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
New test (with -DFORCE_LIBUVC=true), I get 
realsense-viewer
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ ]

Now using ROS I only get one librealsense2 message
roslaunch realsense2_camera rs_camera.launch  enable_color:=false enable_infra1:=false  enable_infra2:=false enable_imu:=false enable_fisheye:=false  enable_depth:=true depth_width:=480  depth_height:=270  depth_fps:=6 enable_pointcloud:=true
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ ]

Thanks for your help
I close this issue

XXLbot, 2019-03-06:
Thanks for your help

closed 2019-03-06 by XXLbot

Eslamrady42, 2020-06-17:
Can you please tell me how did you solved this problem?