realsenseai/realsense-ros #651: Wheel Odometry Relevant API

manomitbal, 2019-03-04:
Hi,

Regarding the statement as posted : 

> Important Notice: For wheeled robots, odometer input is a requirement for robust and accurate tracking. The relevant APIs will be added to librealsense and ROS/realsense in upcoming releases. Currently, the API is available in the underlying device driver.

Is there a timeline to this?

Thank you!

ibaranov-cp, 2019-03-04:
:+1: 
In the meantime, I will likely fuse data with robot localization, treating the camera as a odom source:
http://docs.ros.org/melodic/api/robot_localization/html/configuring_robot_localization.html

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

atyshka, 2019-03-21:
+1 on this, wheel odometry is very important

doronhi, 2019-03-24:
Work in progress. Made PR#691. It is not yet tested enough. When it is tested and approved it will be included in a release.

doronhi, 2019-03-26:
> 👍
> In the meantime, I will likely fuse data with robot localization, treating the camera as a odom source:
> http://docs.ros.org/melodic/api/robot_localization/html/configuring_robot_localization.html

Did you, in the meantime, manage to fuse data with robot localization?
Does [PR#696](https://github.com/intel-ros/realsense/pull/696), as derived from [issue#694](https://github.com/intel-ros/realsense/issues/694) seems to help?

ibaranov-cp, 2019-03-26:
Hello,

I am preparing a small write-up on my adventures with this, hopefully have it done sometime this week :)
Those linked items directly publish a TF. While that is one solution, it does not fuse data from an odom source like wheels, and will conflict with using robot_localization (because that package will also publish TF and having two TF publishers for the same link is very bad (causes fluctuations between the two values))

closed 2019-05-23 by RealSenseCustomerSupport