realsenseai/realsense-ros #652: An error has occurred during frame callback: Error occured during execution of the processing block!

FENGZHANG123, 2019-03-05:
An error has occurred during frame callback: Error occured during execution of the processing block!

I used one D415 and one D435 at same time, PC: UP2 board, Ubuntu 16.04. ROS kinetic.

FENGZHANG123, 2019-03-05:
Librealsense 2.18.1, intel-ros/realsense 2.2.0

freemanlo, 2019-03-07:
Suggest you can also report this issue to below.
https://github.com/IntelRealSense/librealsense/issues/

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

arunabh1904, 2019-03-12:
+1 Im using 2.17.1 with ros 2.1.2. Trying to stream 3 cameras together.

jdwolk, 2019-03-12:
+1, here are my specs:

Device: Raspberry Pi 3 Model B
OS version: Ubuntu MATE 16.04
Camera: D435 (over USB2)
Firmware: 5.11.1.100
librealsense version: 2.18.1 (unpatched, for USB2 support)
ROS realsense version: 2.2.0
ROS version: kinetic

jdwolk, 2019-03-22:
Scratch that, I'm now also getting these errors w/ D435

jdwolk, 2019-03-26:
Bump

Any status on this? I'm still getting these very regularly on D435. Seems like only 1 out of every 10 messages from running this node are successful; the rest give me this error message

doronhi, 2019-03-28:
Sorry, not update. Still working on it..

arunabh1904, 2019-03-28:
The error persists even after updating to v2.19.1 Librealsense and the latest ros wrapper release v2.2.1
Camera model - d435, ROS Melodic.

jdwolk, 2019-04-02:
@doronhi any clue what might be the cause of this? In which codebase (ros-realsense, librealsense) + where in the code might the issue be? I can dig a bit if I know where to start looking

doronhi, 2019-04-03:
So far, what I came up with is this:
The source of the error is in librealsense, rs_processing.hpp, line 334.
From what I understand, the problem is that the filters don't keep up with the frames, due to computer load. A new image cannot be pushed to the still full queue and therefor it is missing when the filter is called to process it. Then the exception is issued.

jinfagang, 2019-04-09:
@doronhi  I am using librealsense 2.18 which not the newest librealsense also got this error, only when I toggled aligned_depth to true maybe also caused by toggled point cloud to true. But the ros realsense wrapper are master branch.

So how do I solve the issue for now???we are in a production enviroment and stable is the most important thing

jdwolk, 2019-04-09:
@jinfagang not a solve but a workaround I found:

If your project isn't married to an SBC yet you can switch to an [Odroid XU4](https://www.hardkernel.com/shop/odroid-xu4-special-price/) (it says sold out but you can purchase them on Amazon). `librealsense` has [semi-official support](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) for building on the Odroid XU4 via a patch. Despite the 4.14 kernel not being in the list of supported kernels, it works with [this ubuntu mate image](https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/20171212)

I just switched from Raspberry PI 3 B -> Odroid XU4 and the issue went away entirely. Odroid XU4 has an 8 core 2GHz CPU compared to the PI's 4 core 1.2 GHz CPU. I'm guessing I no longer see the errors since the beefier SBC can keep up w/ the frames per @doronhi's comment above

Anecdotally, w/ 1 PI + 2 cams over USB2 I got about a pointcloud every 5-10 seconds in between errors. With Odroid XU4 + 2 cams over USB3 I get about 3-5 pointclouds _per second_ and no errors.

jinfagang, 2019-04-10:
@jdwolk  I am afraid this can work longly. I am test it on my PC which has a i7-8th processor and 16G memory, so I think compute capacity is not the issue.

filipematosinov, 2019-04-26:
> @jinfagang not a solve but a workaround I found:
> 
> If your project isn't married to an SBC yet you can switch to an [Odroid XU4](https://www.hardkernel.com/shop/odroid-xu4-special-price/) (it says sold out but you can purchase them on Amazon). `librealsense` has [semi-official support](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) for building on the Odroid XU4 via a patch. Despite the 4.14 kernel not being in the list of supported kernels, it works with [this ubuntu mate 16.04 image](https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/20171212)
> 
> I just switched from Raspberry PI 3 B -> Odroid XU4 and the issue went away entirely. Odroid XU4 has an 8 core 2GHz CPU compared to the PI's 4 core 1.2 GHz CPU. I'm guessing I no longer see the errors since the beefier SBC can keep up w/ the frames per @doronhi's comment above
> 
> Anecdotally, w/ 1 PI + 2 cams over USB2 I got about a pointcloud every 5-10 seconds in between errors. With Odroid XU4 + 2 cams over USB3 I get about 3-5 pointclouds _per second_ and no errors.

I am using a RPi with 2 cameras but I am just getting segmenation faults randomly(using one or 2 cameras). I am looking for alternatives for a RPi with USB3 support that provide no problems with realsense. When you say Ordroid XU4 has no errors it performs perfectly with 2 cameras? Have you tried with 3/4 cameras?

OrHirshfeldFFR, 2019-07-29:
I got a similar error with d435 and librealsense2 2.16.1 on ubuntu 16.04 64bit on a very strong PC machine.
i'm taking photos from few cameras in parallel.

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

Hi @FENGZHANG123

Do you still have the trouble with the latest ros wrapper and librealsense? I don't observe this issue with below command on ros 2.2.7 with the setup on NUC + Ubuntu 16.04 + Kinetic.
$ roslaunch realsense2_camera rs_rgbd.launch camera:=cam_1 serial_no:=<D415 serial number> filters:=pointcloud
$ roslaunch realsense2_camera rs_rgbd.launch camera:=cam_2 serial_no:=<D435 serial number> filters:=pointcloud

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @FENGZHANG123 , @OrHirshfeldFFR,

Is the issue clarified on your platform? Do you try the latest librealsense?
Do you further observe the CPU loading and usb throught impact with multiple cameras using "top" and "usbtop" command?

closed 2019-12-05 by RealSenseCustomerSupport

OrHirshfeldFFR, 2019-12-29:
> Hi @FENGZHANG123
> 
> Do you still have the trouble with the latest ros wrapper and librealsense? I don't observe this issue with below command on ros 2.2.7 with the setup on NUC + Ubuntu 16.04 + Kinetic.
> $ roslaunch realsense2_camera rs_rgbd.launch camera:=cam_1 serial_no:= filters:=pointcloud
> $ roslaunch realsense2_camera rs_rgbd.launch camera:=cam_2 serial_no:= filters:=pointcloud

Sorry for the delay I was on vacation 
I actually don't use the Realsense ROS wrapper, I use librealsense by itself alongside ROS.
As far as I observe there was no issue with CPU or USB overloading, I checked it using "top" and "usbtop" as you suggested
Which librealsense version should solve this problem?

Thank you

CliffSpy74, 2020-10-05:
@jdwolk I apologize for responding to a closed thread, but in all of my scouring I have been unable to find anyone else who mentions implementing ROS-kinetic-realsense2-camera on an ODROID XU-4.  Would you mind sending an outline of how you went about doing this build? I am getting some interesting errors that you seemed to be able to avoid; thanks!