realsenseai/realsense-ros #653: Can frame of pointcloud be aligned to color_frame or depth_frame?

QiuYilin, 2019-03-05:
I found that the frame of pointcloud is camera_depth_optical_frame. But I do the calibration with librealsense api and there is no depth_optical_frame to choose. I have to unify the frames.

doronhi, 2019-03-07:
What tool are you running? What are the options that you **can** choose from?

**depth_optical_frame** refers to the coordinate system in which the point cloud is built in. It is the inner coordinate system of infra1 sensor. There is no sensor called "depth_optical_frame" that you can calibrate.
By using calibration tools you may enhance the quality of the depth calculated and by that receive a better pointcloud.

Is that helpful?

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

QiuYilin, 2019-03-11:
> What tool are you running? What are the options that you **can** choose from?
>
> **depth_optical_frame** refers to the coordinate system in which the point cloud is built in. It is the inner [...]
In fact, I forgot to say it's eye-to-hand calibration. I found that the two frames are same. Thank you for your answer.

closed 2019-03-12 by QiuYilin