realsenseai/realsense-ros #654: Dynamic reconfigure parameters namespace

pavloblindnology, 2019-03-05:
With current implementation
https://github.com/intel-ros/realsense/blob/36dd73aae6bc1b654ad085393eac279339d7f5a0/realsense2_camera/src/base_realsense_node.cpp#L233

all dynamic_reconfigure parameters parameters reside in root nodelet namespace.
This is quite inconvenient, because with multiple cameras we need to use multiple nodelets.
Otherwise there will be names' conflict.
More reasonable is to publish them in camera namespace, by using
`ros::NodeHandle nh1(nh.getNamespace()+"/"+module_name);`
instead.

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

RhysMcK, 2019-05-31:
I also had to make a similar change changes to run all camera under the same nodelet manager.

doronhi, 2019-06-12:
Is #702 tackles this issue?

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@pavloblindnology Can you feedback if https://github.com/IntelRealSense/realsense-ros/pull/702 works for this issue? Thanks!

pavloblindnology, 2019-06-18:
@RealSenseCustomerSupport Yes, #702 fixes this issue.

closed 2019-06-19 by RealSenseCustomerSupport