realsenseai/realsense-ros #658: depth_exposure and other cfg parameters gone

thanujadax, 2019-03-07:
Hi,
In the release 2.1.2, there used to be a folder called `cfg` that contained some dynamically reconfigurable parameters including `depth_exposure` which I found quite useful sometimes. Unfortunately the entire `cfg` directly has been removed in the subsequent releases and I don't see anywhere I can set any parameter that's related to `exposure` anymore.  What was actually the reason for removing these parameters and could you please suggest a workaround to achieve the same result as tuning `depth_exposure`? 
Thanks

eirikaso, 2019-03-07:
don't know if this will help you, but you can use

> rosrun rqt_reconfigure rqt_reconfigure 

This will open a GUI, but if you want something to set exposure at launch I don't know. Haven't tested it yet

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

thanujadax, 2019-03-07:
Thanks. Found a work around to set the parameter at launch, by adding the following call to dynamic reconfigure:
```
      <node name="$(anon param_setter)" pkg="dynamic_reconfigure" type="dynparam"
        args="set Stereo_Module 'Exposure' 1" />
```

closed 2019-03-13 by thanujadax