realsenseai/realsense-ros #660: Dynamic transform support again

pavloblindnology, 2019-03-07:
In previous version of this library, a solution to ROS's multiple latched publishers problem existed #221 #169.
In the current version only static transforms exist, which again returns the problem of multiple cameras configuration running within a single nodelet manager.
Any plans to return the old solution?

RealSenseCustomerSupport, 2019-08-16:
----------------------------------------------

@pavloblindnology After checked #169, I saw Doron suggest you provide PR for this. Are you still working on this? Thanks!

pavloblindnology, 2019-08-16:
@RealSenseCustomerSupport It's almost done. I think I'll finally find the time next week to finalize everything.

pavloblindnology, 2019-08-23:
@RealSenseCustomerSupport Created PR #902

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

@pavloblindnology Thanks very much!

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@pavloblindnology  https://github.com/IntelRealSense/realsense-ros/pull/902 merged to development branch now. Thanks again for your contribution! Will close this accordingly.

closed 2019-09-23 by RealSenseCustomerSupport