realsenseai/realsense-ros #663: D435i: cannot run rs_camera.launch properly

yubaoliu, 2019-03-08:
yubao@yubao-Z370M-S01:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/yubao/.ros/log/339a4e2c-4136-11e9-909c-309c23e1152f/roslaunch-yubao-Z370M-S01-4511.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yubao-Z370M-S01:43333/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
[... 288 lines cut ...]
0x2910cc0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x2425d70) ): Attempt to set a screen on a child window.

yubao@yubao-Z370M-S01:~$ rostopic list 
/ImuFilter/parameter_descriptions
/ImuFilter/parameter_updates
/camera/imu
/camera/rgb/image_rect_color
/diagnostics
/example/another_odom
/example/pose
/example/twist
/imu/data
/imu/mag
/initialpose
/move_base_simple/goal
/odometry/filtered
/rosout
/rosout_agg
/rtabmap/odom
/set_pose
/tf
/tf_static
/voxel_cloud

[image]

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi yubaoliu,

Assuming that you followed the instructions and installed all ROS dependencies and also needed RS SDK.
Which FW your D435i has? And what version of ROS wrapper you have?
Do you have a D435 working with the ROS wrapper?

I tried a D435i w/ FW 5.11,1,100, and it has no issue with rs_camera.launch when using ROS wrapper 2.2.2 + RS SDK 2.19.1 (off Ubuntu 16.04 w/ kernel 4.15.0.46).
And also no issue with D435i w/ FW5.10.13.

Thanks!

yubaoliu, 2019-05-18:
Thanks for your help.
This error disappeared when I reconfigure my ubuntu environment. Maybe some libs are conflicted on my machine.
I recommend using Docker to avoid environmental problems.

RealSenseCustomerSupport, 2019-05-20:
----------------------------------------------

Hi yubaoliu,

Thanks for the updates and feedback.
Glad to hear that the issue is resolved.

Thanks!

closed 2019-05-27 by RealSenseCustomerSupport

hustchenli, 2020-11-04:
> Thanks for your help.
> This error disappeared when I reconfigure my ubuntu environment. Maybe some libs are conflicted on my machine.
> I recommend using Docker to avoid environmental problems.

I have the same problem. Could you please explain how to reconfigure ubuntu environment? I have tried deleting all build-files and rebuild it. But unfortunately it doesn't work.