realsenseai/realsense-ros #664: Mismatched API

CoreyCazes, 2019-03-08:
I was developing with my real sense cameras and everything was working fine. Now whenever I run a launch file I get an error saying "Mismatched API". It says it was compiled with API 2.19.0 but the application was compiled with 2.18.0. I've seen other posts with the same issue but I was unable to resolve it. I'm inexperienced so could someone help me fix this, I'd greatly appreciate it.

doronhi, 2019-03-10:
It seems like you upgraded librealsense to 2.19.0 but did not recompile realsense2_camera as well.
You should try to recompile your verstion of realsense2_camera or, better yet, upgrade reasense2_camera to the latest release (2.2.1).

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@CoreyCazes Did you get chance to try as Doronhi suggested? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@CoreyCazes Did you get through the issue? Could you please update? Thanks!

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@CoreyCazes No response got from your side for weeks. Will close the issue now. Please feel free to create new issues if you still have questions/issues. Thanks!

closed 2019-06-17 by RealSenseCustomerSupport