realsenseai/realsense-ros #666: Can the camera/color/depth/pointcloud message be converted to organized pointCloud?

QiuYilin, 2019-03-11:
I tried to get the coordinate value of a specific point.
```
> void pointCouldCallback(const sensor_msgs::PointCloud2::ConstPtr& point_cloud_msg)
> {
>   //cout<<"pointCloud2 (header):" << point_cloud_msg->header <<endl;
>   std::cout << "x" << u << std::endl;
>   std::cout << "y" << v << std::endl;
>   int pcl_index;
>   //std::cout << "msg_height = " << point_cloud_msg->height << std::endl;
>   
>   pcl::PointCloud<pcl::PointXYZ> point_pcl;
>   pcl::fromROSMsg(*point_cloud_msg, point_pcl);
>   pcl_index = (v*640) + u; 
>   std::cout<< "pcl_index" << pcl_index <<std::endl;
>   std::cout << "(x,y,z) = " << point_pcl.at(pcl_index) << std::endl;
>  
>   
>   std::cout << "cloud: width = " << point_pcl.width << " height = "<<point_pcl.height << std::endl;
>   
> BOOST_FOREACH (const pcl::PointXYZ& pt, point_pcl.points)
> 
>   
>  {
>     
>        camera_x = pt.x;
>      camera_y = pt.y;
>       camera_z = pt.z;
>      printf ("\t(%f, %f, %f)\n", pt.x, pt.y, pt.z);
>       std::cout << "x  = " << camera_x <<" y = "<< camera_y <<" z = " <<camera_z<< std::endl;
>    }
```
But I found that the pointcloud I have got is unorganized and even not static like this
```
cloud: width = 299018 height = 1

cloud: width = 299018 height = 1

cloud: width = 299036 height = 1
```
It seems I am using a wrong method.  What's the right way?

QiuYilin, 2019-03-12:
I used camera/color/depth/point and it is filtered. The right organized topic is /camera/depth_registered/points which is generated by rs_camera/rgbd.launch

closed 2019-03-12 by QiuYilin

mzahana, 2019-03-25:
@QiuYilin I am trying to get the portion of pointcloud that corresponds to a region of interest indicated by a bounding box in the RGB image. Do you know a way to achieve this?
I am using D435. Any hints are appreciated.

Thanks.

pringithub, 2019-06-03:
@mzahana I am trying to do the same thing. Did you happen to get it to work

mzahana, 2019-06-03:
There are two ways I know. Either you get the organized pointcloud and extract the ROI you need, or you calculate the xy of the points of interest using the correponding depth values manually.

aditya2592, 2019-06-07:
I get the following error when trying to use rs_rgbd.launch :

```
ResourceNotFound: rgbd_launch
```

aditya2592, 2019-06-07:
This needs to be installed for it to work  :  

```
sudo apt-get install ros-kinetic-rgbd-launc
```