realsenseai/realsense-ros #669: D435 - USB SCP overflow,1.55236e+12,Error,Hardware Error

jdwolk, 2019-03-12:
Device: Raspberry Pi 3 Model B
OS version: Ubuntu MATE 16.04 
Camera: D435 (over USB2)
Firmware: 5.11.1.100
librealsense version: 2.18.1 (unpatched, for USB2 support)
ROS realsense version: 2.2.0
ROS version: kinetic

Hi!

When I run `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` I get lots of warnings that say the following:
```
[ WARN] [1552356094.393717336, 194.518000000]: Hardware Notification:USB SCP overlfow,1.55236e+12,Error,Hardware Error
```
I get topics showing up in `rqt` but no Pointcloud2 data seems to be appearing in `rviz` on the `/camera/depth_registered/points` as expected.

Also, `librealsense` seems to be working ok as `realsense-viewer`, `rc-depth`, etc, are working.

Thanks in advance for any help!

jdwolk, 2019-03-13:
I was able to find a workaround with the following:

```
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_infra1:=false enable_infra2:=false
```

My guess is I was getting the `USB SCP overlfow` error because too much data was attempting to be trasnfered over USB 2. By disabling infrared I think I took some of the pressure off so now no overflow :man_shrugging:

jdwolk, 2019-05-16:
bump

Still seeing the same thing when I try running higher resolutions and framerates on Odroid XU4

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@jdwolk Could you please provide more details about the higher resolution and FPS you tried? As you're using USB2, the bandwidth is limited which compared with USB3. If possible, you can have a try with USB3 for higher resolution and FPS. Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@jdwolk Any update from your side? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@jdwolk Any other questions about this ticket?

pradeeshenator, 2019-06-15:
I saw a similar problem when using t265. How can I solve for that?

chenbo1173, 2019-06-16:
I have the same problem

chenbo1173, 2019-06-16:
[image]
[image]
please help me with the picture problem .thank you

chenbo1173, 2019-06-16:
Hardware Notification:USB SCP overflow,1.56068e+12,Error,Hardware Error

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@chenbo1173 Did you also see this issue Raspberry Pi 3? If so, then as told in previous comment, the bandwidth of USB2 is limited which compared with USB3. The default launch files enable depth/IR1/IR2/RGB streamings which is very stress for USB2 connection case. Please disable some streaming to try again and share your result. Thanks!

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@chenbo1173 Any other questions about this ticket? If no, will close this accordingly.

closed 2019-07-05 by RealSenseCustomerSupport

MaidenOfTheMain, 2019-10-18:
I'm having the same issue, and it prevents me from using the AR_Marker utility properly.

QuarTerll, 2020-06-03:
same question and wait for a answer @RealSenseCustomerSupport

AndresGalaviz, 2020-09-12:
I'm having the same issue on a USB3 port, any resolution?

patrickw135, 2021-02-15:
Same issues running on Rpi 4 on USB 3 (!!!) connector:

`[WARN] [1613385028.678995219] [RealSenseCameraNode]: Hardware Notification:USB SCP overflow,1.61339e+12,Error,Hardware Error`

when running:  
`ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true device_type:=d435`

after installing [Foxy Version](https://github.com/IntelRealSense/realsense-ros/tree/foxy#step-2-install-librealsense2) but replacing Step 2 with [this installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md).

RealSenseSupport, 2021-03-23:
@QuarTerll @MarcoDotIO @AndresGalaviz @patrickw135 Please type "lsusb -t" to check USB connection is really recognized as USB3. Sometimes the connection will be recognized as USB2 if you use USB2 cable.

huangwen0907, 2021-05-08:
I have the same issue,I have modify and rebuild the code.  when this occur, do the hardware reset, this problem can be fixed.

huangwen0907, 2021-05-08:
see this link https://github.com/huangwen0907/realsense/commit/dc1ab20aeae8a72f77bf24e164ca93e1c0fb8a3a

mvoelk, 2022-02-21:
I had similar issues, after hardware reset and firmware update via realsense-viewer, the errors are gone.