realsenseai/realsense-ros #670: The distance got from pointcloud is inaccurate

QiuYilin, 2019-03-12:
I subscribed the topic "/camera/depth_registered/points"  and Multiply z by 0.1 ， but the data is smaller than the right distance about 5cm. Is the unit right or there is something else I shoud do ?

doronhi, 2019-03-13:
There is an issue with the scale for SR300. You can see the progress in [PR#605](https://github.com/intel-ros/realsense/pull/605)
I'm not sure yet if it implies to the pointcloud but I think you should multiply by 0.125.
Also, 10 cm is very close to the lower limit of the camera. I suggest you make your testings at around 30-50 (cm) if possible.

QiuYilin, 2019-03-13:
> There is an issue with the scale for SR300. You can see the progress in [PR#605](https://github.com/intel-ros/realsense/pull/605)
> I'm not sure yet if it implies to the pointcloud but I think you [...]

And what's the scale of x and y.   I put the target almost in the center but it shows 0.10 0.13 . How can I confirm the scale by myself ?

closed 2024-03-29 by MartyG-RealSense