realsenseai/realsense-ros #672: D435 Auto exposure not stable during startup

thanujadax, 2019-03-13:
We have observed some failures of auto exposure  (release 2.2.0 librealsense 2.18.1) that seems to happen during startup. Basically it is adjusting the exposure during startup and then gets stuck with the wrong exposure. IR image looks over exposed. 
Unchecking and re-checking the dynamic reconfigure didn't work, but restarting helped.
So short term fix is checking the IR image after starting bringup and restart if needed. Any ideas how to fix this?

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@thanujadax What camera are you using and what's firmware version of the camera? Did you still get the same issue with the latest ROS version 2.2.5?

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@thanujadax Did you still need support for this? Looking forward to your update. Thanks!

thanujadax, 2019-06-13:
Hi, I didn't encounter this problem afterwards. We can close this. Thanks!

closed 2019-06-13 by thanujadax