realsenseai/realsense-ros #679: T265 calibration

abylikhsanov, 2019-03-14:
Hi,
I am running librealsense 2.19.1 and when I use ROS i get uncalibrated values for the camera. Should it be like that?

schmidtp1, 2019-04-19:
Hi @abylikhsanov, please update to the latest version and you will receive camera intrinsics:
```
$ rostopic echo /camera/fisheye1/camera_info -n 1
header: 
  seq: 4470
  stamp: 
    secs: 536577153
    nsecs: 375714660
  frame_id: "camera_fisheye1_optical_frame"
height: 800
width: 848
distortion_model: "plumb_bob"
D: [-0.0021744559053331614, 0.03760552033782005, -0.036404311656951904, 0.005936861969530582, 0.0]
K: [285.6466979980469, 0.0, 424.5740051269531, 0.0, 286.68499755859375, 408.1993103027344, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [285.6466979980469, 0.0, 424.5740051269531, 0.0, 0.0, 286.68499755859375, 408.1993103027344, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```
Please notice that the distortion model is Kannala Brandt 4 (and not plumb_bob as stated above). This will be fixed.

abylikhsanov, 2019-05-20:
Hi, any update regarding the fix of the distortion model?

liushuya7, 2019-06-17:
How to modify the distortion model to 'Kannala Brandt 4' instead of 'plumb_bob'?

canberkgurel, 2020-05-11:
@abylikhsanov No update as of 11 May 2020. But the above fix will probably work.

helenol, 2020-05-26:
Pinging on this! Could we please just add the following line? base_realsense_node.cpp, line 1710:
```C++
    if (intrinsic.model == RS2_DISTORTION_KANNALA_BRANDT4) {
      _camera_info[stream_index].distortion_model = "equidistant";
    } else {
      _camera_info[stream_index].distortion_model = "plumb_bob";
    }
```

**Kannala Brandt 4 *is* equidistant :)**

I'll make a PR at some point soon.

doronhi, 2020-06-02:
> Pinging on this! Could we please just add the following line? base_realsense_node.cpp, line 1710:
>
> ```c++
>     if (intrinsic.model == RS2_DISTORTION_KANNALA_BRANDT4) { [...]

I think you are right. I'll happily accept a PR from you.

helenol, 2020-06-03:
Great, made a PR for this: https://github.com/IntelRealSense/realsense-ros/pull/1225 :)

closed 2020-09-21 by RealSenseSupport