realsenseai/realsense-ros #680: Possible Mismatch: RGB Data Type of Published Colored Point Cloud

sahandy, 2019-03-14:
Name       |    Configuration
---------------- | -------------------
Operating System | Ubuntu 16.04 Xenial
Kernel Version   | 4.14.12-rt10
ROS              | Kinetic
RealSense-Ros    | 2.1.4
librealsense     | 2.19.1
Sensor           | D435

I would like to subscribe to the colored pointcloud topic `/camera/depth/color/points` to perform some post-processing steps on that colored point cloud. To achieve so, I run the [standard launch](https://github.com/intel-ros/realsense/blob/development/realsense2_camera/launch/rs_camera.launch) file with the `pointcloud` filter:

```bash
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
```
My subscriber looks like the following:
```c++
ros::Subscriber cloud_sub = nh.subscribe<sensor_msgs::PointCloud2>("/camera/depth/color/points", 1, cloudCB);
```
and here is my callback function:
```c++
void cloudCB(const sensor_msgs::PointCloud2ConstPtr& cloud_msg)
{
  typedef typename pcl::PointCloud<pcl::PointXYZRGB> ColoredCloud;
  ColoredCloud::Ptr cloud(new ColoredCloud);
  pcl::fromROSMsg(*cloud_msg, *cloud);
  // ... do something on `cloud` ...
  // ...
}
```
However, I receive the following warnings from PCL and the launch file, respectively:
```bash
Failed to find match for field 'rgb'.
```
```bash
[ WARN] [1552574519.108911426]: No stream match for pointcloud chosen texture Process - Color
```
I took a look at a message header published over the `/camera/depth/color/points` topic:
```console
$ watch -n2 rostopic echo /camera/depth/color/points
Every 2.0s: rostopic echo /camera/depth/color/points                                                                                                                                                           Thu Mar 14 16:22:41 2019

header:
  seq: 314
  stamp:
    secs: 1552576961
    nsecs: 568551949
  frame_id: "camera_depth_optical_frame"
height: 1
width: 142339
fields:
  -
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  -
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  -
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  -
    name: "rgb"
    offset: 16
    datatype: 6
    count: 1
is_bigendian: False
point_step: 20
row_step: 2846780
```

Refering to [`sensor_msgs::PointCloud2::PointField`](http://docs.ros.org/api/sensor_msgs/html/msg/PointField.html), the rgb data type is set to 6 (a.k.a `uint8 UINT32`). However, it seems like `fromROSMsg()` expects rgb data type of 7 (a.k.a `uint8 FLOAT32`). I realized this by simply creating a dummy pointcloud, using `toROSMsg()` and checking the PointField of that message.

I also tried the following which hides the conversion step, but the result was the same:
```c++
ros::Subscriber cloud_sub = nh.subscribe<pcl::PointCloud<pcl::PointXYZRGB>>("/camera/depth/color/points", 1, cloudCB);
...
...
void cloudCB(const pcl::PointCloud<pcl::PointXYZRGB>& cloud)
{ }
```

Is the above observation correct? If so, is there a way that I could receive a colored point cloud properly using ROS?

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@sahandy Sorry for late response. Did you still have issue with the latest ROS version? Looking forward to your update. Thanks!

0xJustin, 2019-09-26:
I am also having this issue in realsense-ros 2.2.8

sahandy, 2019-10-07:
@RealSenseCustomerSupport 
thanks for reaching out. Yes the problem still persists. Currently, we are using realsense-ros 2.2.8.

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@sahandy @0xJustin Thanks for the update! Will look into this.

JTShepard, 2019-10-18:
I found a workaround by modifying line 2023 in `base_realsense_node.cpp` (v2.2.8) from:

`msg_pointcloud.point_step = addPointField(msg_pointcloud, format_str.c_str(), 1, sensor_msgs::PointField::UINT32, msg_pointcloud.point_step);`

to:

`msg_pointcloud.point_step = addPointField(msg_pointcloud, format_str.c_str(), 1, sensor_msgs::PointField::FLOAT32, msg_pointcloud.point_step);`

Hope it helps!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@JTShepard Thanks very much for share!
@sahandy Is the workaroud working for you? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@sahandy Any update from your side? Thanks!

sahandy, 2019-11-13:
We tested the suggestion made by @JTShepard and it worked. The point cloud ROS message is now prepared and published correctly.

@RealSenseCustomerSupport Is it possible to have this fix in the official repo, as a pull request or something?

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@JTShepard  @sahandy Glad to hear that works. Yes, you're welcome to submit PR for the fix.

closed 2019-12-02 by RealSenseCustomerSupport

jonra1993, 2019-12-03:
Hi @RealSenseCustomerSupport. I am using version2.2.9 but problem still persists. Do you have any recommendation?
[image]
[image]