realsenseai/realsense-ros #681: Unable to get x,y,z position in /camera/depth/color/points topic

Levi568, 2019-03-16:
From issue 633, I changed the topic in rviz to the one @doronhi suggests. I can see that it is a poincloud2 sensor message. However, I need to get the real x,y,z position (like float values) of the points. I tried pcl conversion but it turned out to get the points that the sensor recieved, which I don't need. Is there any way to convert the pointcloud2 points to x,y,z values?

Thank you

jlysnantel, 2019-03-20:
I am still exploring but what I got right now is 
```
import sensor_msgs.point_cloud2 as pc2
from sensor_msgs.msg import PointCloud2, PointField
```
where pc2 is what you want to use to manipulate your pointcloud2 object.
And here's my code where I received a pointcloud2 from a service 

```
Points = self._points_proxy(1).m_data
        for point in pc2.read_points(Points, field_names = ("x", "y", "z"), skip_nans=True):
            pt_x = point[0]
            pt_y = point[1]
            pt_z = point[2]
            print(" x : %f  y: %f  z: %f", point[0],point[1],point[2])
```

I think the values are in meter

Levi568, 2019-03-21:
What is your "self._points_proxy(1).m_data"? I set it to sensor message and turns out the values really close to zero

jlysnantel, 2019-03-21:
It's a proxy to acces the service that I created that  exposed the topic /camera/depth/color/points. If you are wondering what type it is, it's a pointcloud2

Levi568, 2019-03-30:
Does it suppose to have many zero values? I am confused of the x,y,z values because when I subscribe the /camera/depth/color/points topic, there are many zeros .

jlysnantel, 2019-03-30:
Or you aiming to a wall or multiple object ? If there is object in front of other I think that the "shadow" behind the object will be zeroes. Try to aim at a wall and only print values that are not zero so you can validate that the Z coordinate is good

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@jlysnantel Thanks very much for the input!
@Levi568 Sorry for late response. Do you still have questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@Levi568 Could you please update? Thanks!

Levi568, 2019-09-29:
@RealSenseCustomerSupport Sorry for late reply. I don't have any question now. It works well on me.

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@Levi568 That's great! Thanks very much for the update!

closed 2019-10-12 by RealSenseCustomerSupport

Alex-Beh, 2020-07-19:
@Levi568 I managed to get the z real world distance from depth image and the unit is in mm. I tried to use the compute the x & y distance using the intrinsic matrix and z value however the result is wrong.

May I know how to get the x & y distance in real world from the realsense camera?