realsenseai/realsense-ros #685: Is it possible to have the depth from the topic  /camera/aligned_depth_to_color/image_raw realsense D435

jlysnantel, 2019-03-19:
I am working with the Realsense D435 and this intel-ros/realsense Package. I am working with pictures, so I made a Service that published a picture from certain topic of the realsense. 
I am trying to find the depth of a pixel with the topic /camera/aligned_depth_to_color/image_raw. 

I do not understand how the encoding is 16U1C when the data array type is a uint8 (default data type of a sensor_msgs/Image type).

When I look at the image in rviz I see   
[image]

And when I published it with my code 
`
def _get_aligned_depth(self):
        print("Get depth")
        AlignedDepth = self._aligned_depth_proxy(
        cv_image = self._bridge.imgmsg_to_cv2(AlignedDepth, "16UC1")
        cv2.imwrite('/tmp/pic/alignDepth.png',cv_image)
        print(type(cv_image))
`
I see this 
[image]

I am not sure where to look when I use this package and his topic is there documentation or is it the same as the Realsense?

doronhi, 2019-03-20:
You get an image encoded as 16UC1.
If you want to see it on screen you have to convert it to 8bit image.

jlysnantel, 2019-03-20:
I finaly understood that I have to merge two cell from the data array to form the form to distance on  16 bit .

closed 2019-03-21 by jlysnantel