realsenseai/realsense-ros #686: D435i acceleration vectors display improperly in RViz

hello-ck, 2019-03-19:
Hello, 

I encountered an issue with the display of acceleration vectors from the D435i in RViz. Specifically, the issue arose when I used ros-melodic-rviz-imu-plugin to visualize the acceleration vectors published to the /camera/accel/sample topic. 

I was able to correct the visualization by creating a node that transformed the topic in the following way: 

        x = input.linear_acceleration.x
        y = input.linear_acceleration.y
        z = input.linear_acceleration.z
        output.linear_acceleration.x = -y
        output.linear_acceleration.y = -z
        output.linear_acceleration.z = x

This correction resulted in the values for the axes matching the values displayed with the realsense-viewer. 

The details for the system I used follow: 

D435i with firmware Version: 5.11.1.100
Intel RealSense Viewer v2.19.1
ROS Wrapper 2.0 for Intel RealSense Devices (build 2.2.1)
librealsense2 version: 2.19.1-0~realsense0.631
Ubuntu 18.04
ROS Melodic

This issue may be related to #649 .

doronhi, 2019-03-20:
Sounds like an issue but I can't recreate. I am using kinetic.
When using the plug I get the following error:

> rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
Aborted (core dumped)

Seen anything like that?
Was it working before you changed the code? What is the Fixed Frame that you use?
The fix should not be hard, I would just like to have a way of checking myself.

hello-ck, 2019-03-20:
Yes. I encountered a similar error and worked around it by changing the options for the plugin in RViz. The following screenshot shows the options I am currently using.

[image]

I also encountered timing issues that you might see. I reduced the timing errors by lowering the sample rate for the accelerometer, and eventually downsampling the rate in my custom node that performs the transformation. 

```
    <arg name="accel_fps" value="63"/>`
```

In addition, I set the following parameters, although I'm unsure if they made a difference. 

```
    <arg name="enable_sync" value="true"/>
    <arg name="initial_reset" value="true"/>
```

More information about the uncorrected topic follows:

```
$ rostopic echo /camera/accel/sample
header: 
  seq: 317505
  stamp: 
    secs: 1553094371
    nsecs: 899543524
  frame_id: "camera_accel_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -6.47238874435
  y: -7.03136777878
  z: 1.08853816986
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
```

More information about the transformed topic follows: 

```
$ rostopic echo /camera/accel/sample_corrected
header: 
  seq: 17686
  stamp: 
    secs: 1553094538
    nsecs: 849687099
  frame_id: "camera_accel_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: 7.01175451279
  y: -1.08853816986
  z: -6.45277547836
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
```

Thank you for looking into this.

hello-ck, 2019-04-22:
Hello, 

I'm following up, since I'm not aware of any changes since I reported this issue. 

At this point, I only use my sample_corrected version throughout my code, since it provides a properly oriented acceleration vector with respect to my URDF.

My main concern is that this will eventually be corrected and break things. I believe this issue relates to #726.

I'd appreciate any guidance you can provide on the expected timing for resolving this issue. 

Thank you.

doronhi, 2019-05-21:
Sorry abut the delay. Please checkout #778. It seems to solve the issue.

hello-ck, 2019-05-25:
Thank you for working on this and notifying me about the update. 

I just tried out the new release with my D435i. I'm still encountering a problem. When I visualize the /camera/accel/sample topic directly, the acceleration vector does not point in the correct direction in RViz. However, changing the frame_id of the messages from camera_accel_frame to camera_accel_optical_frame appears to result in the correct visualization. 

Maybe the coordinate frames being broadcast to tf by /camera/realsense2_camera_manager have an error? Below, I've included a screenshot from the frames.pdf generated by running `rosrun tf2_tools view_frames.py` followed by relevant transforms.  

[image]

```
$ rosrun tf tf_echo /camera_link /camera_accel_frame
At time 0.000
- Translation: [-0.012, -0.006, 0.005]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```
```
$ rosrun tf tf_echo /camera_link /camera_accel_optical_frame
At time 0.000
- Translation: [-0.012, -0.006, 0.005]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]
```
```
$ rosrun tf tf_echo /camera_accel_frame /camera_accel_optical_frame
At time 0.000
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]
```

I won't be available to look at this again until June 3. 

Best wishes.

doronhi, 2019-05-26:
@hello-ck ,
Thanks! That was simply a bug. You are correct that the frame_id should have been accel_optical_frame.
Fixed in #784.

closed 2019-06-12 by doronhi

hello-ck, 2019-07-01:
It works! Thank you!