realsenseai/realsense-ros #687: T265 TF is incorrect

ibaranov-cp, 2019-03-19:
TF values for T265 are incorrect, they have the left and right frames exactly co-located, and camera link to optical frame is backwards.

doronhi, 2019-03-20:
T265 has different coordinate system then the D400 series.
You can compare 
D400: 
[image] 
T265:
[image]

ibaranov-cp, 2019-03-20:
Yep, understood. However the camera frames for left and right should still not be identical.

doronhi, 2019-03-20:
That is correct. The extrinsics are not available yet by the device. The TF values will be fixed automatically, as soon as the firmware is fixed.

schmidtp1, 2019-04-25:
This had been fixed. What can be seen here are the two fisheye optical frames together with the T265 body frame (in the center).
[image]
Thanks for filing the issue, @ibaranov-cp!

closed 2019-04-25 by schmidtp1

ibaranov-cp, 2019-04-25:
Great, thanks!

SainaRez, 2019-06-14:
@doronhi @schmidtp1 What confuses me is that if we treat the camera_pose_frame diagram in rviz the same as the one in the image, the camera's align on the y axis in rviz while in the image they align on the x axis. 
Could you please provide a diagram with the camera and the imu_frame fisheye frame and body frame similar to the one below from vins-fusion:
[image]

schmidtp1, 2022-10-12:
> @doronhi @schmidtp1 What confuses me is that if we treat the camera_pose_frame diagram in rviz the same as the one in the image, the camera's align on the y axis in rviz while in the image they [...]

sorry, i just saw this again now... i think the confusion comes from the different reference/body frames used in ROS/rviz and T265/librealsense. T265 uses a coordinate system as depicted in the first diagram ("VR coordinates") with y pointing up, z aligned with the optical axis and x with the baseline of the cameras, that is exposed by librealsense. Here in the ROS wrapper there is a transformation that rotates the frame to follow ROS/robotics convention with z pointing up etc. as can be seen in the screenshot from rviz. hope this helps.