realsenseai/realsense-ros #688: URDF and mesh for D435i and T265

gaoethan, 2019-03-20:
Currently there is no corresponding URDF to describe the D435i and T265, Where to find these kinds of information to allow the right TF transformation in ROS ?

atyshka, 2019-03-21:
D435i is exactly the same as D435 which has a URDF provided

gaoethan, 2019-03-22:
okay, that's good. But D435i has an IMU embedded than D435, how about the description regarding the IMU in D435i? thanks

SteveGrehl, 2019-07-22:
Hi, is there a *mesh/urdf* for the T265 yet? I'd like to integrate the camera into our robot and those files are missing here imo.
Integrating the D435 is straight forward, hoping to implement the T265 too.

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

Hi @SteveGrehl, @gaoethan

The information in the following may help
https://github.com/IntelRealSense/librealsense/issues/4584

closed 2019-09-18 by RealSenseCustomerSupport

saikishor, 2019-10-23:
@SteveGrehl @gaoethan Do you guys have the URDF of the

saikishor, 2019-11-12:
Can someone verify if the frames in the URDF file of T265 here are correct?
https://github.com/pal-robotics/realsense_gazebo_plugin/blob/kinetic-devel/urdf/t265.urdf.xacro