realsenseai/realsense-ros #689: DDynamic Reconfigure conflicting versions

v-lopez, 2019-03-20:
The version of ddynamic_reconfigure used in this project is based on this repository: https://github.com/awesomebytes/ddynamic_reconfigure

Unfortunately, that repository's code is a fork from: https://github.com/pal-robotics/ddynamic_reconfigure

But the author, license and maintainers were removed, we have requested a fix for this here: https://github.com/awesomebytes/ddynamic_reconfigure/issues/5

Also, the whole API has changed, but the name hasn't, so now the derived package is totally incompatible with the original, but has the same name.

We are releasing the original ddynamic_reconfigure on the ROS build farm for our users, and I'm afraid this may conflict with the users of this repository.

We offer to extend our package to be able to be used as a replacement here, and can contribute to fixing #620 and #609 .

Otherwise I guess that the derived package's name should be changed. 

I'd like to hear your opinion on this, before we start working on extending our API and migrating your code.

doronhi, 2019-03-20:
@v-lopez 
The reason the realsense repository has a copy of ddynamic_reconfigure in it was the difficulties and annoyance for users with submodules.
If you manage to release your package as a debian package it will be best.
Otherwise, assuming it's possible, I will happily sync the realsense copy with yours as soon as yours is ready.

v-lopez, 2019-03-20:
I'm in the process of releasing the current version right now, are you targeting kinetic and melodic? Any other distro?

We'll make the necessary changes to use our version of ddynamic_reconfigure and test it with the realsense we have in the office, which is how we found out about this conflict.

doronhi, 2019-03-20:
Officially we only support kinetic so it's good.
If, by the end of your tests, you could issue a PR it will be great.
Otherwise, let me know and I'll do the sync as soon as possible.

v-lopez, 2019-03-21:
https://github.com/intel-ros/realsense/pull/693

veimox, 2019-05-20:
Hi! What is the argument now to keep a semi-fork of `ddynamic_reconfigure` in this repo? Could we consider removing it?

doronhi, 2019-05-21:
No argument. It will be removed as soon as possible.

patelmiteshn, 2019-05-31:
Has this been fixed if not what a way around to this ?

doronhi, 2019-06-02:
> Has this been fixed if not what a way around to solve this problem

The latest development branch of realsense2_camera has the ddynamic_reconfigure library removed already. What problem do you need to work around?

patelmiteshn, 2019-06-04:
Hi doronhi,

The realsense2_camera still has ddynamic_reconfigure referenced in thee CMakeLists.txt.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/CMakeLists.txt#L18

doronhi, 2019-11-10:
Hi @patelmiteshn ,
realsense2_camera depends on ddynamic_reconfigure. Therefor it appears in it's CMakelists.txt file.
Is this ok? Can we close the issue?

v-lopez, 2019-11-11:
Yeah, I think so.

closed 2019-11-11 by v-lopez