realsenseai/realsense-ros #694: T265 frame_id-s for odometry message and TF

pavloblindnology, 2019-03-21:
Hi.
I'm trying to fuse T265 odometry with other localization data using robot_localization.
For this I need odometry message with correct frame_id-s specified (e.g. `header.frame_id` = "map" and `child_frame_id`="camera_link").
But current version publishes TF with the same frame_id-s - cluttering with robot_localization output.
So, to avoid this T265's TF `camera_link` need to have an option to be detached from `camera_pose_frame`. And odometry message to have `child_frame_id` different from TF's counterpart.
Thanks.

pavloblindnology, 2019-03-22:
As a more general question.
I don't want to feed T265 with wheel odometry to improve VSLAM.
I want to feed T265 odometry to robot_localization to fuse with all other odometry sensors - wheel, GPS, other IMU-s following [REP105](http://www.ros.org/reps/rep-0105.html).
Do you see any benefits in doing it your way (feeding wheel odometry to T265)?

doronhi, 2019-03-24:
> Hi.
> I'm trying to fuse T265 odometry with other localization data using robot_localization.
> For this I need odometry message with correct frame_id-s specified (e.g. `header.frame_id` = "map" and [...]

Not sure I understand. You can change the frame id's with launch params `odom_frame_id` and `pose_frame_id`. Does that help?

doronhi, 2019-03-24:
> As a more general question.
> I don't want to feed T265 with wheel odometry to improve VSLAM.
> I want to feed T265 odometry to robot_localization to fuse with all other odometry sensors - wheel, GPS, [...]

The obvious benefit is saving on your machine's CPU. Other that that, it's a matter of which algorithm works best for you.
There is a disclaimer that without the additional information of the wheel odometry the device cannot be reliable over time. Since this device is meant to give reasonable performance as a stand alone and without additional computation power from the host, it offers the option to give it the additional information.

pavloblindnology, 2019-03-25:
Of course, saving host's resources is obvious. Another benefit is that T265's relocalization should work better with better odometry.
But there are disadvantages too. Like, in order to fuse with GPS or other odometry sources (e.g., 2nd IMU), or to follow [REP-105](http://www.ros.org/reps/rep-0105.html) (which is usually the case for outdoor robots at least), I need to use smth like [robot_localization](http://wiki.ros.org/robot_localization) anyway.

pavloblindnology, 2019-03-25:
> Not sure I understand. You can change the frame id's with launch params `odom_frame_id` and `pose_frame_id`. Does that help?

I mean that, in order to follow [REP-105](http://www.ros.org/reps/rep-0105.html) and use [robot_localization](http://wiki.ros.org/robot_localization), one needs to have Realsense's TF part and odometry message published with different frame_id-s. Otherwise Realsense's TF clutters with that of `robot_localization`. Or there should be an option to turn off publishing of `odom_frame_id` -> `pose_frame_id` TF link. Or even better - to have both options, and an option to detach `odom_frame_id` -> `pose_frame_id` TF link from the rest of T265's TF frames.

doronhi, 2019-03-25:
I've issued [PR696](https://github.com/intel-ros/realsense/pull/696)
I did not yet try to combine the output of T265 into another localization node so I hope you can check this minimal correction and see if it enables your work.

closed 2019-07-23 by RealSenseCustomerSupport