realsenseai/realsense-ros #695: T265 stops to send pose

JozefRSTU, 2019-03-22:
When start realsense node with rs_t265.launch file I am able to view pose data using rostopic echo, T265 is tracking movement but after some times, when T265 is in rest with no movement, rostopic is receiving only "nan" instead of numbers in translation vector and orientation quaternion. I am not sure what exactly is the condition to put camera into this state, but it falls into this state every time I try it but it takes different time. Intel realsense node is still publishing opse topic at rate approx. 200Hz but no real data available. To recovery from this state I have to physically switch off and on again the T265 into USB port. As I am new in this area I am not sure what should I check for better description of this issue but hope someone can me point in right direction. Behavior is same on my laptop running Ubuntu 16.04 with patched (realsenselib related patch) kernel and on Jetson TX2 (Ubuntu 16.04) without patch. ROS version is kinetic.

closed 2019-05-14 by JozefRSTU

lcluz, 2019-08-14:
Hi,
This is happening to me. But I don't understand if there is a solution for it or what's the cause. I am using librealsense 2.25 and realsense-ros 2.2.8

manomitbal, 2019-09-25:
@JozefRSTU  Were you able to find a solution to this problem?

neilyoung, 2019-10-16:
Having the same issue. Runs for an hour, then stop. No indication, no error :(