realsenseai/realsense-ros #699: D435 roslaunch error:Frame metadata isn't available!

carol007, 2019-03-26:
Hi,friends!

**Environments**
Device: Dell xps15
OS version: Ubuntu  16.04 lts
Ubuntu kernel: 4.15.0-46-generic
Camera: D435 
Firmware: 5.11.4 (latest development release version)
librealsense version: SDK2.0 
Build 2.19.1
ROS realsense version: 2.2.1
ROS version: kinetic

**My issue:**
  I use USB3.0 and can run realsense-viewer successfully.
  But when I enter code `roslaunch realsense2_camera rs_camera.launch`  ,I get warn  [ WARN] [1553594396.682607742]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME).   Moreover,after I open the rviz, nothing could't been shown in the relative windows. I tried to Click "Add" and "By topic" , pointcloud2 cannot be found. The situation is same with image stream and depth stream. All the things don't work in Rviz.  Can any friend help me?

**Details:**
roslaunch realsense2_camera rs_camera.launch
... logging to /home/carol/.ros/log/e8295c7c-4fad-11e9-a9fd-9cb6d0b9f3a7/roslaunch-awesome-5027.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://awesome:42407/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 65 lines cut ...]
[ INFO] [1553594396.427157354]: Expected frequency for depth = 30.00000
[ INFO] [1553594396.447328987]: Expected frequency for infra1 = 30.00000
[ INFO] [1553594396.464790763]: Expected frequency for infra2 = 30.00000
[ INFO] [1553594396.482403450]: Expected frequency for color = 30.00000
[ INFO] [1553594396.501577739]: setupStreams...
 26/03 05:59:56,503 WARNING [140077112743808] (sensor.cpp:366) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1553594396.505088969]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553594396.505223817]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553594396.505342068]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 26/03 05:59:56,506 WARNING [140077112743808] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 26/03 05:59:56,507 WARNING [140077112743808] (sensor.cpp:366) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1553594396.511740402]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553594396.512960004]: insert Depth to Stereo Module
[ INFO] [1553594396.512988229]: insert Color to RGB Camera
[ INFO] [1553594396.512999627]: insert Infrared to Stereo Module
[ INFO] [1553594396.513038020]: insert Infrared to Stereo Module
[ INFO] [1553594396.529810826]: num_filters: 0
[ INFO] [1553594396.529849277]: SELECTED BASE:Depth, 0
[ INFO] [1553594396.531031377]: RealSense Node Is Up!
[ INFO] [1553594396.531047843]: Setting Dynamic reconfig parameters.
 26/03 05:59:56,680 WARNING [140076182578944] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
**[ WARN] [1553594396.682607742]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)**
 26/03 05:59:56,696 WARNING [140076323432192] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ INFO] [1553594396.704691789]: Done Setting Dynamic reconfig parameters.
[image]

doronhi, 2019-03-26:
I'm not sure about the warning. Does it show every time, even when disconnecting and reconnecting the camera?
The problem with the point cloud is that you should specify param. type:
`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
You will need to change the `fixed frame` in RViz to see it.
However, I can't see why you don't get images. Can you add the log after the point you already added?
What can you see when typing: `rostopic hz /camera/color/image_raw` ?

carol007, 2019-03-27:
@doronhi  Thank you for your reply,Sir.
------------------------------------------------------------------------------------
**Does it show every time, even when disconnecting and reconnecting the camera?**
Ans: Yes,this situation always happen when I dis and reconnect the camera and enter `roslaunch realsense2_camera rs_camera.launch` 
[image]
---------------------------------------------------------------------------------------
**The problem with the point cloud is that you should specify param.**
Ans: Yes. I follow your code ` roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
this terminal immediately shows that
[image]

then I open a new terminal to enter rviz. Nothing happened as following.

Then I check other closed issues and follow their suggestions to Click "Add" and "By topic", I cannot find the choice "PointCloud2", so I have to randomly choose a image stream or depth stream to test whether it works to show. Unfortunately, NO image received by Rviz...
[image]
[image]
-------------------------------------------------------------------------------------------
Can you add the log after the point you already added?
Ans: Sorry,cause I am a coding&ROS newcomer.I don't quiet understand how to add log after the point.
-------------------------------------------------------------------------------------------
What can you see when typing: rostopic hz /camera/color/image_raw ?
Ans: When I directly enter your code in the /root or ~/catkin_ws, something wrong. Then I realize I should launch a node with `roslaunch realsense2_camera rs_camera.launch`  then enter `rostopic hz /camera/color/image_raw `, I get some message as followed.
[image]
--------------------------------------------------------------------------------------------
Because I don't know how to change the fixed frame shown in RVIZ with red color Font, maybe my operation is wrong to some degree.  Moreover, I guess Rviz essentially doesn't receive my data stream. Is this related to my  environment ? That is too latest..
Firmware: **5.11.4** 
Build **2.19.1**  
ROS realsense version: **2.2.1**

I will try to degrade FW to 5.11.1  and Build 2.19.0 to test. Would you mind recommending me available environment solution?  Thank you!

carol007, 2019-03-27:
@doronhi  Thank you for your reply,Sir.
-------------------------------------------------
Does it show every time, even when disconnecting and reconnecting the camera?

Ans: Yes,this situation always happen when I dis and reconnect the camera and enter `roslaunch realsense2_camera rs_camera.launch` 

[image]

--------------------------------------------------------------------------------------------------------------------------
The problem with the point cloud is that you should specify param.

Ans: Yes. I follow your code ` roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
this terminal immediately shows that
[image]

then I open a new terminal to enter rviz. Nothing happened as following.

[image]

Then I check other closed issues and follow their suggestions to Click "Add" and "By topic", I cannot find the choice "PointCloud2", so I have to randomly choose a image stream or depth stream to test whether it works to show. Unfortunately, NO image received by Rviz...

[image]

[image]

------------------------------------------------------------------------------------------------------------------
Can you add the log after the point you already added?

Ans: Sorry,cause I am a coding&ROS newcomer.I don't quiet understand how to add log after the point.

---------------------------------------------------------------------------------------------------------------------------------
What can you see when typing: rostopic hz /camera/color/image_raw ?

What can you see when typing: rostopic hz /camera/color/image_raw ?
Ans: When I directly enter your code in the /root or ~/catkin, something wrong. Then I realize I should launch a node with `roslaunch realsense2_camera rs_camera.launch`  then enter `rostopic hz /camera/color/image_raw `, I get some message as followed.

[image]

-----------------------------------------------------------------------------------------------------------------------
Because I don't know how to change the fixed frame shown in RVIZ with red color Font, maybe my operation is wrong to some degree.  Moreover, I guess Rviz essentially doesn't receive my data stream. Is this related to my  environment ? That is too latest..
Firmware: 5.11.4 
Build 2.19.1  
ROS realsense version: 2.2.1

I will try to degrade FW to 5.11.1  and Build 2.19.0 to test. Would you mind recommending me available environment solution?  Thank you!

carol007, 2019-03-28:
@doronhi    Unfortunately, after I degrade my SDK from 2.19.1 to 2.19.0 to fit the recommended requirements of ROS2.2.1, the camera still cannot open with warning UVC metadata payloads not available and Frame metadata isn't available!

doronhi, 2019-04-01:
Did you install the kernel patches?
from https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md:
```
Verify that the kernel is updated :
modinfo uvcvideo | grep "version:" should include realsense string
```
Try to run:
```
sudo apt-get update
sudo apt-get install librealsense2-dkms
```

Jiang1206, 2019-04-04:
> 你安装了内核补丁吗？
> 来自https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md：
> 
> ```
> Verify that the kernel is updated :
> modinfo uvcvideo | grep "version:" should include realsense string
> ```
> 尝试运行：
> 
> ```
> sudo apt-get update
> sudo apt-get install librealsense2-dkms
> ```

Jiang1206, 2019-04-04:
I also encountered this problem, I made changes according to your method, but the problem remains. Also, I use the D430 depth module with no color, no point cloud is related to no color. 
When I open rviz and try to add a pointcloud, there is a second error:"[ WARN] [1554379179.581464881]: No stream match for pointcloud chosen texture Process - Color
"

doronhi, 2019-04-04:
@Jiang1206 , the problem you describe comes from the fact that the default texture is color. You can change it with rqt_reconfigure or by adding the following parameter to the launch command: `pointcloud_texture_stream:=RS2_STREAM_ANY`

Jiang1206, 2019-04-06:
What do you mean by entering the command at the terminal: "roslaunch realsense2_camera demo_pointcloud.launch pointcloud_texture_stream:=RS2_STREAM_ANY"? But this still has no image in rviz, and it prompts an error: "too many values to unpack
The traceback for the exception was written to the log file”

doronhi, 2019-04-07:
@carol007 , according to `rostopic hz` the images do arrive so I would check what I see with `rqt_image_view` now. Did you try the kernel patches?
@Jiang1206 , `demo_pointcloud.launch` doesn't include the `pointcloud_texture_stream` option. I meant `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY`
Also, can you confirm with `rostopic list` and `rostopic hz` that the topics for pointcloud and images are published? 
Which app prompts the error "too many values..."?

Grandzxw, 2019-10-31:
> ## @doronhi Thank you for your reply,Sir.
>
> Does it show every time, even when disconnecting and reconnecting the camera?
>
> Ans: Yes,this situation always happen when I dis and reconnect the camera [...]
Have you done it successfully?

RealSenseCustomerSupport, 2020-02-08:
----------------------------------------------

Is this still reproducible in the latest ROS/librealsense release?
Thank you.

closed 2020-02-11 by RealSenseCustomerSupport

lavanya463, 2020-02-13:
Hi, 
I have a similar problem when I disconnect and reconnect it solves... But it is really difficult to do it every time manually.
I am using the following command to get a point cloud
`roslaunch realsense2_camera rs_rgbd.launch enable_infra1:=false enable_infra2:=false`

and many times my `rostopic hz /camera/depth_registered/points` produce no new messages.
Can you please help me or provide some path to solve this?
usually this happens after working with the realsense for some time (6-7 hours).

I am using realsense D415.

doronhi, 2020-02-13:
Sorry for losing track of your issue guys. I do hope you managed after all.
@lavanya463, Do you mean that after 6-7 hours you suddenly get "Frame metadata isn't available"?
In any case, this seems like a different issue.
I think you should open a new issue and describe the problem. 
Also, when you do, it would help if you state what FW you use. What kernel, librealsenses version, machine?
An issue that happens after several hours needs as many details as possible to try to recreate..

sbdjyl, 2020-06-15:
Hi,friends!
I face the same problem .It cames from  intel nuc  on my UAV .
The onboard computer have intel i7 7567u ,and the ubuntu environments is same as yours.
I got librealsense-2.9.0 or 2.9.1 for my D435i,when i run 

roslaunch realsense2_camera rs_camera.launch

it tells my Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

Have you done it successfully?Can you help me?
Thanks

lavanya463, 2020-06-15:
@doronhi Sorry for the late reply. I wasn't able to reproduce the same issue with my new PC. I will let you know here if I face the same issue. Thank you.

@sbdjyl Normally Frame metadata isn't available can be solved by ignoring some outputs of realsense. For example I run " roslaunch realsense2_camera rs_camera.launch enable_infra1:=false enable_infra2:=false etc "
I hope this is helpful.

Thanks