realsenseai/realsense-ros #700: Pointcloud Corresponding to Region of Interest in RGB Image

mzahana, 2019-03-26:
Hi.
I am using D435 and I would like to get the portion of pointcloud that corresponds to a region of interest (ROI) in the RGB image indicated by a bounding box. What is a possible way to achieve that?

Thanks.

mzahana, 2019-03-26:
@doronhi @icarpis Any hints are appreciated. Thanks.

mzahana, 2019-04-04:
I found a possible solution.

I used ```rs_rgbd.launch``` (which requires ```rgbd_launch``` pkg) to get an organized pointcloud. Then, I could match the region of interest determined in the RGB frame to the one in the organized pointcloud ```camera/depth_registered/points```.

closed 2019-04-04 by mzahana

khgapa, 2020-11-04:
@mzahana Could you please help me how to solve the problem of extracting points from the ROI of an RGB image?