realsenseai/realsense-ros #701: Question about calibration matrices

eirikaso, 2019-03-26:
When using the ROS wrapper for Realsense D435, I can see the calibration matrix on the /camera/"camera"/camera_info topic. However I can see different calibration parameters when I use the "Intel.Realsense.CustomRW" to read off the calibration matrix that is stored on the camera. 

I was under the impression that the calibration parameters published on the /camera/"camera"/camera_info topic were being read off the camera and would therefore be the same.

So the questions are:
Are both these calibration parameters utilized and added to each other? 
I would like to recalibrate the camera under water using the ROS/opencv stereo calibration tool. Should I create a new calibration matrix for publication on the /camera/"camera"/camera_info topic, or should I create a calibration matrix and write it to the camera using "Intel.Realsense.CustomRW"?

Here are the calibration parameters from "Intel.Realsense.CustomRW"

> Calibration parameters from the device:
>   resolutionLeftRight: 1280 800
> 
>   FocalLengthLeft: 640.590088 640.183533
>   PrincipalPointLeft: 637.286316 402.818481
>   DistortionLeft: -0.055629 0.063853 -0.000203 -0.000965 -0.020792
> 
>   FocalLengthRight: 642.577454 642.286743
>   PrincipalPointRight: 648.905396 406.992188
>   DistortionRight: -0.056637 0.067119 -0.000290 -0.001725 -0.022515
> 
>   RotationLeftRight: 0.999975 -0.002564 -0.006569
>                      0.002552 0.999995 -0.001754
>                      0.006574 0.001738 0.999977
>   TranslationLeftRight: -49.850437 -0.109076 0.017740

Calibration parameters published on /camera/"camera"/camera_info

> header: 
>   seq: 869
>   stamp: 
>     secs: 1552302510
>     nsecs: 879911038
>   frame_id: "camera_infra1_optical_frame"
> height: 800
> width: 1280
> distortion_model: "plumb_bob"
> D: [0.0, 0.0, 0.0, 0.0, 0.0]
> K: [639.0849609375, 0.0, 644.4653930664062, 0.0, 639.0849609375, 364.2340393066406, 0.0, 0.0, 1.0]
> R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
> P: [639.0849609375, 0.0, 644.4653930664062, 0.0, 0.0, 639.0849609375, 364.2340393066406, 0.0, 0.0, 0.0, 1.0, 0.0]
> binning_x: 0
> binning_y: 0
> roi: 
>   x_offset: 0
>   y_offset: 0
>   height: 0
>   width: 0
>   do_rectify: False

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@eirikaso Sorry for late response. Which ROS version did you use when get the issue? Do you get chance to try the latest SDK to see if still issue to you? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@eirikaso Could you please update? Thanks!

eirikaso, 2019-09-30:
Hi
I think I was using kinetic. 
I ended up making my own ros node to publish the raw uncalibrated image stream (which is not published by your ros wrapper) and my own calibration matrix. I then used stereo matching in opencv to create a disparity map and do measurements.

I have also written the calibration matrix directly to the camera, but have not checked the accuracy of measurements using the distortion map generated on the camera.

What I think is that since the calibration parameters written to the camera is already added to the published image, the published calibration parameters does not contain any distortion data since no further processing is required to obtain the calibrated image.

closed 2019-10-01 by eirikaso

wangcongrobot, 2021-03-19:
> When using the ROS wrapper for Realsense D435, I can see the calibration matrix on the /camera/"camera"/camera_info topic. However I can see different calibration parameters when I use the "Intel.Realsense.CustomRW" to read off the calibration matrix that is stored on the camera.
> 
> I was under the impression that the calibration parameters published on the /camera/"camera"/camera_info topic were being read off the camera and would therefore be the same.
> 
> So the questions are:
> Are both these calibration parameters utilized and added to each other?
> I would like to recalibrate the camera under water using the ROS/opencv stereo calibration tool. Should I create a new calibration matrix for publication on the /camera/"camera"/camera_info topic, or should I create a calibration matrix and write it to the camera using "Intel.Realsense.CustomRW"?
> 
> I see that distortion and rotation only is set in the parameters on the camera.
> 
> Here are the calibration parameters from "Intel.Realsense.CustomRW"
> 
> > Calibration parameters from the device:
> > resolutionLeftRight: 1280 800
> > FocalLengthLeft: 640.590088 640.183533
> > PrincipalPointLeft: 637.286316 402.818481
> > DistortionLeft: -0.055629 0.063853 -0.000203 -0.000965 -0.020792
> > FocalLengthRight: 642.577454 642.286743
> > PrincipalPointRight: 648.905396 406.992188
> > DistortionRight: -0.056637 0.067119 -0.000290 -0.001725 -0.022515
> > RotationLeftRight: 0.999975 -0.002564 -0.006569
> > 0.002552 0.999995 -0.001754
> > 0.006574 0.001738 0.999977
> > TranslationLeftRight: -49.850437 -0.109076 0.017740
> 
> Calibration parameters published on /camera/"camera"/camera_info
> 
> > header:
> > seq: 869
> > stamp:
> > secs: 1552302510
> > nsecs: 879911038
> > frame_id: "camera_infra1_optical_frame"
> > height: 800
> > width: 1280
> > distortion_model: "plumb_bob"
> > D: [0.0, 0.0, 0.0, 0.0, 0.0]
> > K: [639.0849609375, 0.0, 644.4653930664062, 0.0, 639.0849609375, 364.2340393066406, 0.0, 0.0, 1.0]
> > R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
> > P: [639.0849609375, 0.0, 644.4653930664062, 0.0, 0.0, 639.0849609375, 364.2340393066406, 0.0, 0.0, 0.0, 1.0, 0.0]
> > binning_x: 0
> > binning_y: 0
> > roi:
> > x_offset: 0
> > y_offset: 0
> > height: 0
> > width: 0
> > do_rectify: False

The camera_info published by ROS is not equal to the calibration parameters from "Intel.Realsense.CustomRW". Actually, it is equal to the calibration parameters from "rs-enumerate-devices -c". Both of them have zero D parameters, which means the image is not raw data but rectified image.

So I guess the calibration parameters from "Intel.Realsense.CustomRW" are used to rectify the raw image because the D is not zero. If we rewrite the calibration parameters using "Intel.Realsense.CustomRW", the inner parameters will be changed.

Is it correct? @RealSenseCustomerSupport @MartyG-RealSense

MartyG-RealSense, 2021-03-19:
@wangcongrobot  The camera's **Y16** format is not rectified as it is used for camera calibration.  The rest of the camera's image formats are rectified.

The 400 Series cameras set coefficients to zero for reasons described in the link below.  

https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

The color stream coefficient of the **D455** camera model has been observed to be non-zero though.

https://github.com/IntelRealSense/realsense-ros/issues/1580

wangcongrobot, 2021-03-19:
D is not zero when run `/usr/bin/Intel.Realsense.CustomRW -r`:
```bash
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0B3A
  Device name: Intel RealSense D435I
  Serial number: 947522071820
  Firmware version: 05.12.11.00

Calibration parameters from the device:
  resolutionLeftRight: 1280 800

  FocalLengthLeft: 641.472656 640.786011
  PrincipalPointLeft: 632.790161 402.550018
  DistortionLeft: -0.055414 0.059363 0.000124 0.000641 -0.019018

  FocalLengthRight: 634.501038 633.512146
  PrincipalPointRight: 638.121033 394.077850
  DistortionRight: -0.056966 0.063801 -0.000390 0.000565 -0.020186

  RotationLeftRight: 0.999914 -0.002408 -0.012855
                     0.002378 0.999994 -0.002350
                     0.012861 0.002319 0.999915
  TranslationLeftRight: -49.938873 -0.016061 -0.388207

  HasRGB: 1

  resolutionRGB: 1920 1080

  FocalLengthColor: 1369.939331 1368.062622
  PrincipalPointColor: 971.746338 537.572571
  DistortionColor: 0.000000 0.000000 0.000000 0.000000 0.000000
  RotationLeftColor: 0.999986 -0.002521 -0.004678
                     0.002537 0.999991 0.003455
                     0.004669 -0.003467 0.999983
  TranslationLeftColor: 15.062349 0.089593 0.062225
```

wangcongrobot, 2021-03-19:
I know the last five parameters of `Left Intrinsics` is `D/coeff`, but not sure about the first four parameters.

MartyG-RealSense, 2021-03-20:
Hi @wangcongrobot   Intel have provided me with the meanings of the Camera Calibration interface's values, which are read from the depth calibration table.

Left Intrinsics: Left camera intrinsic data, normalized
Right Intrinsics: Right camera intrinsic data, normalized
World To Left Rotation: The inverse rotation of the left camera
World To Right Rotation: The inverse rotation of the right camera

wangcongrobot, 2021-03-24:
The left Intrinsics should be equal to this:
```
  FocalLengthLeft: 641.472656 640.786011
  PrincipalPointLeft: 632.790161 402.550018
  DistortionLeft: -0.055414 0.059363 0.000124 0.000641 -0.019018
```
https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-802943243

Where is the RGB camera intrinsic?

MartyG-RealSense, 2021-03-25:
You can query the calibration values of the camera with **rs-enumerate-devices -c**.  The tool will list the calibration details for all of the stream profiles in the terminal and then you can scroll up through the listings until you find the color intrinsics for each color resolution supported by that camera.

The example below is taken from my D435i.

[image]