realsenseai/realsense-ros #704: [realsense2] release to kinetic/ubuntu 16.04

SteveMacenski, 2019-03-28:
Per #410 with 5 people also agreeing this needs to happen...

Please stop closing tickets that are open and needing action just because there's been no movement. Issues are issues regardless if anyone's touched them.

MartyG-RealSense, 2019-03-28:
I understand your concern at cases being closed after a period of inactivity.  It is necessary to close cases though to make it easier to keep track of issues that are still clearly awaiting a response (sometimes the creator of a message will stop responding to further requests for information, for example).    

If you need to highlight an issue that has been closed but you still need assistance with, you are welcome to create a new case yourself and provide a link to the closed one for the reference of the Intel team members.

SteveMacenski, 2019-03-28:
Yeah but closing a ticket because no one on the Intel side has actually fixed the issue at hand isn't an acceptable way of running an issue tracker in my opinion. That system is closing legitimate issues and concerns because no one fixed the underlying problem in over a year.

No one fixing the issue isn't the same thing as the issue no longer appearing.

MartyG-RealSense, 2019-03-28:
Some issues take longer than others to fix.  This is especially so in cases where the fix needs to go in the firmware.  It is necessary to be far more careful with additions to firmware than it is with SDK updates due to the risk of 'bricking' the camera with a faulty firmware. 

Intel do track unresolved issues, I promise, and maintain a list of unresolved issues in the release notes of both the SDK and the firmware.

SteveMacenski, 2019-03-28:
That's understandable, but similarly Intel has chosen to support this driver as part of its choice to enter the robotics space and have active ROS support, I don't see how an issue from Microsoft, Google, etc being a reason for tickets summary closed while still exhibiting the behavior. I don't think many of the tickets closed fall into this category, and even this one which just involves a 20 minute bloom release we've been waiting on for nearly a year. 

I totally appreciate the fact that you all are working on making a ROS driver, but closing issues that you haven't had time to touch isn't an acceptable way to run an open source project effecting literally hundreds of robotics companies.

MartyG-RealSense, 2019-03-28:
Thanks so much for your constructive feedback.   Your passion demonstrates how much you care.  :).  There isn't much else I could add to the debate as I am not an Intel employee, so hopefully one of the Intel guys on this forum such as @doronhi can continue it with you.  Thanks again!

doronhi, 2019-04-01:
Sorry guys, I can't see how much I can contribute. You are both perfectly right.
The technical support team does wonders, helping filter out the issues. 
Some issues get closed by mistake.
Some issues take way too long to solve.
Some features take way too long to be incorporated.

This specific issue, creating a release package, should not have been closed. It is very important in my opinion.

AlistairSymonds, 2019-05-07:
Agreed, in the meantime I've bee using this in a makefile, is there any better work around? (especially given I've got manually update the url for each version)
```
ifeq ($(wildcard ./src/realsense*/.),)
	wget https://github.com/intel-ros/realsense/archive/2.2.3.tar.gz
	tar -C ./src -zxvf 2.2.3.tar.gz
	rm 2.2.3.tar.gz
	@echo "$(G_COLOR) Realsense ROS and system packages installed $(NO_COLOR)"
else
	@echo "$(G_COLOR) Realsense ROS package source already exists, skipping $(NO_COLOR)"
endif

doronhi, 2019-05-12:
No technical reason why kinetic is the latest supported version. Just technicalities..

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@SteveMacenski This should be resolved. Any other question about this? Looking forward to your update. Thanks!

closed 2019-11-07 by RealSenseCustomerSupport

SteveMacenski, 2019-11-07:
This is still a problem, please stop closing tickets without action. 

Issues are issues regardless if anyone’s working on it.