realsenseai/realsense-ros #705: Pipeline syncer is not included in 2.19 version and is creating issue during build.

Brolin56, 2019-03-28:
The issue is not fixed in the updated version i.e, 2.19.0. So I have to manually place the lines from this [merge request](https://github.com/intel-ros/realsense/pull/452/files) to make my code work

doronhi, 2019-03-28:
I'm not sure what the issue is? The PR is merged in realsense2_wrapper.
Could you please elaborate?

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

@Brolin56 As dorohni replied the PR is already merged. Could you please double check? Thanks!

RealSenseCustomerSupport, 2019-08-07:
----------------------------------------------

@Brolin56 Sorry that we didn't get reply from you for weeks. Will close the issue accordingly. Please feel free to create another one if you still have issues/questions. Thanks!

closed 2019-08-07 by RealSenseCustomerSupport