realsenseai/realsense-ros #706: Multiple T265 cameras

Yu-LanChang, 2019-03-28:
My system is Ubuntu 16.04, and I have some problem when I try to get data from two T265 cameras.

After connecting two T265 cameras to PC, I ran the following commands from different terminals.

```

roslaunch realsense2_camera rs_t265.launch camera:=cam_1 serial_no:=<serial number of the first camera>
roslaunch realsense2_camera rs_t265.launch camera:=cam_2 serial_no:=<serial number of the second camera>
```

I can get the data from the first camera (cam_1). However, the information from the second camera (cam_2) cannot be acquired. The second terminal only shows 
```
[ INFO] [1553711925.173709452]: Initializing nodelet with 8 worker threads.
[ INFO] [1553711925.195283050]: RealSense ROS v2.2.1
[ INFO] [1553711925.195304806]: Running with LibRealSense v2.19.1
[ WARN] [1553711925.204682574]: No RealSense devices were found!
```
Thanks.

doronhi, 2019-04-01:
It is an issue. For now it is not possible to run more then one node handling T265 on the same machine.

Yu-LanChang, 2019-04-01:
Thanks for your reply. And when could I get a new version including this function? Thanks.

francelico, 2019-04-02:
Yes, it would be very helpful to have the capability to launch multiple t265 in the near future.

stino78, 2019-06-11:
Also, right now it is possible to stream data from two t265 units with the Intel RealSense Viewer! So in principle it is possible.

I am not able to do it right now...when I try to start the second unit I get an error: "RuntimeError: Failed to start TM2 camera"

Any tips to do it?

THank you!!

doronhi, 2019-06-11:
It's possible to use two T265 with realsense viewer because the viewer is only 1 process.
With realsense2_camera, each node handles one device. Since a process can either hold the T265 context or not, you can use only one T265 at a time.
The real solution to this problem is replacing the backend for T265 in librealsense. A possible workaround is modifying the function unload_tracking_module in such a way that it will keep the hold of the current process on the device while releasing the context for other processes to search for other devices.
In any case, the solution is in the underlying librealsense2 library.

ivomarvan, 2019-08-02:
I wrote code in Python 3 to test two T265 cameras. It works.

[more_devices_test.py.zip](https://github.com/IntelRealSense/realsense-ros/files/3461264/more_devices_test.py.zip)

stino78, 2019-08-02:
Wonderful! Thank you for doing it!

Agostino

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

Is there anything more needed on this ticket?  Right now we do not have the capability but it seems we have some savvy users (Thanks @ivomarvan) and the python script was made available.

closed 2019-10-04 by RealSenseCustomerSupport

cimbar, 2019-12-02:
Actually, the solution is not related with ros package. Do you plan to implement this feature on the next release?

stino78, 2020-03-08:
Dear Intel Support, 
the wonderful code written by IVOMARVAN is not working anymore with the new release of librealsense 2.33.1.1370.

When I try to run it (Ubuntu 18.04), I have the following runtime error:

_RuntimeError                              Traceback (most recent call last)
<ipython-input-8-6bbf1f461706> in <module>
    182 if __name__ == "__main__":
    183     number_of_experiments = 50
--> 184     serial_numbers =  get_devices_serial_numbers()
    185     sources = [T265CameraSource(serial_number) for serial_number in serial_numbers]
    186 

<ipython-input-8-6bbf1f461706> in get_devices_serial_numbers(device_suffix)
     30     ret_list = []
     31     ctx = rs.context()
---> 32     for d in ctx.devices:
     33         if device_suffix and not d.get_info(rs.camera_info.name).endswith(device_suffix):
     34             continue

RuntimeError: Unable to open device interface_

Can you help me with that?