realsenseai/realsense-ros #707: Depth align to RGB plus post-processing

JeffLee06, 2019-03-28:
Hi,
I am running D435i in ubuntu 16.04 under kinetic ROS, 
Is it possible to run align_depth with post processing filters. 
I have tried to open both in launch/rs_camera.launch but doesn't seem to work.

jlysnantel, 2019-03-31:
Try
```
roslaunch realsense2_camera rs_camera.launch align_depth:=true
```

JeffLee06, 2019-03-31:
Hi, 
Enabling align_depth:=true works perfectly fine, my question was about opening align_depth and filters:=... together. The filters:= options only work on the depth_raw topic

doronhi, 2019-04-01:
The`filters:=` option works on the raw depth and the aligning part takes place afterwards.
In other words, the filtered depth image is aligned to produce the aligned depth image.
Is that behavior not working for you? In that case, can you write down the command line you use?

RealSenseCustomerSupport, 2019-09-04:
----------------------------------------------

Hi @JeffLee06,

Any update? Is your issue resolved per @doronhi comment?

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

Hi @JeffLee06,

If no further question, will

closed 2019-11-11 by RealSenseCustomerSupport