realsenseai/realsense-ros #708: Cannot find package configuration realsense2

omar-ogm, 2019-03-29:
Hi, So while I did install ros wrapper realsense-2.2.0 previously following the instructions here https://github.com/intel-ros/realsense/#installation-instructions, now I cant install the realsense-2.2.1 ros wrapper. 
What I did last time was follow the instructions in the previous url, and since the last version of intel realsense 2.0 SDK (https://github.com/IntelRealSense/librealsense/releases) match the latest version of ros wrapper I did install the ros realsense SDK with apt as explained here (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)

Since now the latest version of the realsense 2.0 SDK is not supported by the latest version of the ros realsense wrapper I have to download a previous one.

So using "apt install" I installed the version needed of the intel realsense SDK. This are the ones that I have installed (apt list --installed | grep realsense):
**********
intel-realsense-dfu/xenial,now 1.0.0 amd64 [installed]
librealsense2/xenial,now 2.19.0-0 realsense0.614 amd64 [installed,upgradable to: 2.19.1-0 realsense0.643]
librealsense2-dkms/xenial,now 1.3.5-0ubuntu1 all [installed]
librealsense2-udev-rules/xenial,now 2.19.0-0  realsense0.614 amd64 [installed,upgradable to: 2.19.1-0 realsense0.643]
librealsense2-utils/xenial,now 2.19.0-0 realsense0.614 amd64 [installed,upgradable to: 2.19.1-0 realsense0.643]
**********
NOTICE that I installed version 2.19.0-0 by doing apt install librealsense2=2.19.0-0.
The only one that was installed using the last version was librealsense2-dkms, since I didnt know which version to use.

This installation seems to work since realsense-viewer app seems to work.

The problem comes when I tried to install the ros realsense wrapper (build 2.2.1):
On Step 3 (https://github.com/intel-ros/realsense/#step-3-install-intel-realsense-ros-from-sources), when doing "catkin_make clean" I get the following error:
**********
CMake Warning at realsense-2.2.1/realsense2_camera/CMakeLists.txt:36 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2"
  (requested version 2.19.0) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at realsense-2.2.1/realsense2_camera/CMakeLists.txt:38 (message):
  
   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
**********
My guess here is that installing a specific version using apt didnt install the SDK properly, and now it cannot find the cmake file.

If thats the case, I would like to know is there is any way to install intel realsense SDK version 2.19.0 using apt.

Otherwise I can try an install version 2.19.0 using source code.

zlacelle, 2019-07-11:
Confirming that librealsense2 is missing the cmake file in the deployed packages in xenial/kinetic.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@omar-ogm Sorry for late response. Could you please try the latest SDK and ROS wrapper and feedback? Thanks!

omar-ogm, 2019-11-22:
I just installed from sources and it works. I guess we can close this

closed 2019-11-22 by omar-ogm

Harsharma2308, 2020-08-07:
Why do I have to install from source?

brianavila72, 2020-12-14:
> Why do I have to install from source?

I have the same exact question. I am using Ubuntu 18.04.5 LTS with Kernel 5.4.0-generic and because of that I cannot install from source (it only supports kernels 4.x), so I installed Intel realsense2 with DKMS package, but when I try to compile my project the Cmakelist cannot find realsense2.

Cmakelist: find_package(RealSense2 REQUIRED)

I am stuck here.