realsenseai/realsense-ros #709: Intrinsic Camera Parameters - D435

jlysnantel, 2019-03-31:
Is there a topic that publish the Intrinsic Camera Parameters so I can use the depth data to calculate the coordinate (x,y,z) of a pixel. I use the D435. I want to calculate the coordinates as stated in https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0?fbclid=IwAR3gogVZe824YUps88Dzp02AN_XzEm1BDb0UbmzfoYvn1qDFb7KzbIz9twU#intrinsic-camera-parameters for the deprojection.

jlysnantel, 2019-04-01:
I now understand that they are under /camera/nameOfYourTypeOfTopicCam/camera_info. 

You can see http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CameraInfo.html To have better information

closed 2019-04-01 by jlysnantel

jinfagang, 2019-04-04:
@jlysnantel  Do you solved your problem?  How to subscribe topic? which topic do you using ? does there any snippets in c++ to do this?

jlysnantel, 2019-04-04:
My tests have not been done yet. But what I have now is the rgb image in the topic `/camera/color/image_raw`. I have the depth of a pixel in the topic `/camera/aligned_depth_to_color/image_raw` and I have the intrinsic parameter of the camera in the topic `/camera/aligned_depth_to_color/camera_info`.

The topic camera_info return an object of type [sensor_msgs/CameraInfo](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CameraInfo.html). As said in the documentation:
<pre>
**Intrinsic camera matrix for the raw (distorted) images.**
     [fx  0 cx]
 K = [ 0 fy cy]
     [ 0  0  1]
 Projects 3D points in the camera coordinate frame to 2D pixel
 coordinates using the focal lengths (fx, fy) and principal point
(cx, cy).
</pre>

Whit all these information we have everything to calculate the [deprojection of a pixel](https://github.com/IntelRealSense/librealsense/blob/5e73f7bb906a3cbec8ae43e888f182cc56c18692/include/librealsense2/rsutil.h#L46).

```
/* Given pixel coordinates and depth in an image with no distortion or inverse distortion coefficients, compute the corresponding point in 3D space relative to the same camera */
static void rs2_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics * intrin, const float pixel[2], float depth)
{
    assert(intrin->model != RS2_DISTORTION_MODIFIED_BROWN_CONRADY); // Cannot deproject from a forward-distorted image
    assert(intrin->model != RS2_DISTORTION_FTHETA); // Cannot deproject to an ftheta image
    //assert(intrin->model != RS2_DISTORTION_BROWN_CONRADY); // Cannot deproject to an brown conrady model

    float x = (pixel[0] - intrin->ppx) / intrin->fx;
    float y = (pixel[1] - intrin->ppy) / intrin->fy;
    if(intrin->model == RS2_DISTORTION_INVERSE_BROWN_CONRADY)
    {
        float r2  = x*x + y*y;
        float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;
        float ux = x*f + 2*intrin->coeffs[2]*x*y + intrin->coeffs[3]*(r2 + 2*x*x);
        float uy = y*f + 2*intrin->coeffs[3]*x*y + intrin->coeffs[2]*(r2 + 2*y*y);
        x = ux;
        y = uy;
    }
    point[0] = depth * x;
    point[1] = depth * y;
    point[2] = depth;
}
```
I find these information there : https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

ravijo, 2019-10-02:
Hi @jinfagang

I made a ROS package to do the same. Please check it out [here](https://github.com/ravijo/ros_openpose).

Cheers!

AndreV84, 2020-08-26:
is it possible to get the camera_intrinsic.json file for d435i from anywhere?

doronhi, 2020-08-26:
I'm not sure what you mean by json file but the intrinsics values are available using the command:
`rs-enumerate-devices -c`

AndreV84, 2020-08-26:
@doronhi Thank you for following up!
I am trying to install open3D
the latter seem to require camera file provided
https://support.intelrealsense.com/hc/en-us/community/posts/360039393953-Highlighting-the-JetScan-3D-scanner-build-project-for-Jetson-Nano
https://github.com/devshank3/JetScan#usage

yuan0623, 2021-05-11:
> My tests have not been done yet. But what I have now is the rgb image in the topic `/camera/color/image_raw`. I have the depth of a pixel in the topic `/camera/aligned_depth_to_color/image_raw` and I have the intrinsic parameter of the camera in the topic `/camera/aligned_depth_to_color/camera_info`.
> 
> The topic camera_info return an object of type [sensor_msgs/CameraInfo](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CameraInfo.html). As said in the documentation:
> 
> **Intrinsic camera matrix for the raw (distorted) images.**
>      [fx  0 cx]
>  K = [ 0 fy cy]
>      [ 0  0  1]
>  Projects 3D points in the camera coordinate frame to 2D pixel
>  coordinates using the focal lengths (fx, fy) and principal point
> (cx, cy).
> Whit all these information we have everything to calculate the [deprojection of a pixel](https://github.com/IntelRealSense/librealsense/blob/5e73f7bb906a3cbec8ae43e888f182cc56c18692/include/librealsense2/rsutil.h#L46).
> 
> ```
> /* Given pixel coordinates and depth in an image with no distortion or inverse distortion coefficients, compute the corresponding point in 3D space relative to the same camera */
> static void rs2_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics * intrin, const float pixel[2], float depth)
> {
>     assert(intrin->model != RS2_DISTORTION_MODIFIED_BROWN_CONRADY); // Cannot deproject from a forward-distorted image
>     assert(intrin->model != RS2_DISTORTION_FTHETA); // Cannot deproject to an ftheta image
>     //assert(intrin->model != RS2_DISTORTION_BROWN_CONRADY); // Cannot deproject to an brown conrady model
> 
>     float x = (pixel[0] - intrin->ppx) / intrin->fx;
>     float y = (pixel[1] - intrin->ppy) / intrin->fy;
>     if(intrin->model == RS2_DISTORTION_INVERSE_BROWN_CONRADY)
>     {
>         float r2  = x*x + y*y;
>         float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;
>         float ux = x*f + 2*intrin->coeffs[2]*x*y + intrin->coeffs[3]*(r2 + 2*x*x);
>         float uy = y*f + 2*intrin->coeffs[3]*x*y + intrin->coeffs[2]*(r2 + 2*y*y);
>         x = ux;
>         y = uy;
>     }
>     point[0] = depth * x;
>     point[1] = depth * y;
>     point[2] = depth;
> }
> ```
> 
> I find these information there : https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

Hi, I have checked the `/camera/aligned_depth_to_color/camera_info`, and I found my distortion model is `plumb_bob`, which is neither "Modified Brown-Conrady Distortion" nor `Inverse Brown-Conrady Distortion`. I think this affects the deproject results because in the code it actually looks for the distortion model. Do you have any advice on this? What is `plumb_bob` equivalent to? 

Best
Yuan

PierreLouisT, 2021-12-30:
Hi everybody, do you know if it is possible to modify those parameters without using Calibration tool ? For example, with a code is it possible to ecrase intrinsics parameters and change them the way we want. I am searching how to correct a sphere cloudpoint. I read some issues where it is adviced to use Dynamic Calibrator and on-chip calibrator but it is not efficient. I think if we force the change of intrinsics we could change the surface of the pointcloud. Do you agree with that ? 
Do you have an idea about it ?

lukistar14, 2022-03-07:
Plumb bob seems to be equivalent to the Brown Conrady distortion model, according to this link: https://calib.io/blogs/knowledge-base/camera-models

Also found these lines of code in `realsense-ros/realsense2_camera/scripts/show_center_depth.py`, 71-74, which support this hypothesis:
```
if cameraInfo.distortion_model == 'plumb_bob':
    self.intrinsics.model = rs2.distortion.brown_conrady
elif cameraInfo.distortion_model == 'equidistant':
    self.intrinsics.model = rs2.distortion.kannala_brandt4
```

vocdex, 2022-09-11:
Hi everyone,
Did anyone calculate the camera intrinsic matrix. Would you please share it here?
Thanks