realsenseai/realsense-ros #712: error of pointcloud after updating to the latest version

QiuYilin, 2019-04-02:
I am using SR300 and there was something wrong about the scale of pointcloud in the past. But after I update to the newest modified version, all the value of coodinate become nan.

I subscribe the  "camera/depth_registered/points", topic.

```
 ros::Subscriber point_cloud_sub =
      nh.subscribe("camera/depth_registered/points", 1, pointCouldCallback);
```

and deal the msgs like this 

```
int u,v;

void darknetCallback(const darknet_ros_msgs::BoundingBoxes::ConstPtr &msg)

{
  obj_class = msg->bounding_boxes[0].Class;
  u = (msg->bounding_boxes[0].xmin + msg->bounding_boxes[0].xmax) / 2;
  v = (msg->bounding_boxes[0].ymin + msg->bounding_boxes[0].ymax) / 2;
}

void pointCouldCallback( const sensor_msgs::PointCloud2::ConstPtr &point_cloud_msg) {

  pcl::PointCloud<pcl::PointXYZ> point_pcl;
  pcl::fromROSMsg(*point_cloud_msg, point_pcl);

  auto pt = point_pcl.at(u,v);
  camera_x = 0.124987*pt.x;
  camera_y = 0.124987*pt.y;
  camera_z = 0.124987*pt.z;
   if (obj_class == target_obj &&(camera_x >= -1.5 ||camera_x<=1.5))
  {
  std::cout << "x  = " << camera_x << " y = " << camera_y
            << " z = " << camera_z << std::endl;
  }
  obj_class = "none";
}
```
pointcloud is like this in rviz
[image]

doronhi, 2019-04-04:
Its a bug. Thank you for informing me.
I just merged PR#717 to fix it.
I'd appreciate if you checked the latest version.

QiuYilin, 2019-04-08:
> Its a bug. Thank you for informing me.
> I just merged PR#717 to fix it.
> I'd appreciate if you checked the latest commit.

In fact, the point cloud  is even more strange. @doronhi

doronhi, 2019-04-10:
Checked again with latest commit (f548ca5e43229d22c75ac3f994f9e4d85debf4d8) on development branch with librealsense v2.20.0, with `rs_rgbd.launch` and `rs_camera.launch filters:=pointcloud`.
Running SR300, firmware 3.21.0.0.
Point cloud in RViz looks fine.
Where do you see the problem?

QiuYilin, 2019-04-10:
Sorry,I found that I stll used the released version and missed the commit. @doronhi

doronhi, 2019-04-15:
Excellent.  Thanks.

closed 2019-04-15 by doronhi