realsenseai/realsense-ros #721: camera_info topics are confusion

jinfagang, 2019-04-09:
There are many camera info topics...

```
/camera/color/camera_info
/camera/aligned_depth_to_color/camera_info 
```

How do I use it if I need to calculate the 3d point from image pixel and depth?

I logged them out, they seems acutally same?

```
/camera/aligned_depth_to_color/camera_info 
header: 
  seq: 1015
  stamp: 
    secs: 1554777396
    nsecs:  93076351
  frame_id: "camera_color_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [924.4299926757812, 0.0, 639.8958740234375, 0.0, 924.4682006835938, 364.4192199707031, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [924.4299926757812, 0.0, 639.8958740234375, 0.0, 0.0, 924.4682006835938, 364.4192199707031, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

/camera/color/camera_info
header: 
  seq: 1428
  stamp: 
    secs: 1554777403
    nsecs: 979297455
  frame_id: "camera_color_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [924.4299926757812, 0.0, 639.8958740234375, 0.0, 924.4682006835938, 364.4192199707031, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [924.4299926757812, 0.0, 639.8958740234375, 0.0, 0.0, 924.4682006835938, 364.4192199707031, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

doronhi, 2019-04-10:
Since the aligned image is projected on the color sensor, camera_info describes color sensor in both topics. They are the same. You can use either one.

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@jinfagang Any other questions about this? Looking forward to your update. Thanks!

closed 2019-09-23 by RealSenseCustomerSupport