realsenseai/realsense-ros #722: D435 difference between software registeration and hardware registeration

qiuchenli, 2019-04-10:
Dear Sir:
I doubt the different between software registeration and hardware registeration , which is shown in rs_rgbd.launch, as below:

To launch Realsense with software registeration (ROS Image Pipeline and rgbd_launch):
    $ roslaunch realsense2_camera rs_rgbd.launch
Processing enabled by ROS driver:
    # depth rectification
Processing enabled by this node:
    # rgb rectification
    # depth registeration
    # pointcloud_xyzrgb generation

To launch Realsense with hardware registeration (ROS Realsense depth alignment):
    $ roslaunch realsense2_camera rs_rgbd.launch align_depth:=true
Processing enabled by ROS driver:
    # depth rectification
    # depth registration
Processing enabled by this node:
    # rgb rectification
    # pointcloud_xyzrgb generation 

For stability， which one is prefered?

FPSychotic, 2019-07-17:
I ask by the same but I would get info of the other settings and how they affect to sw and hw reg.
 A link is enough

adrianramospeon, 2019-07-24:
I also wonder about the performance differences. Is there an advantage (or difference) to generate the pointcloud in the base_realsense_node (with realsense code), instead of using the dept_image_proc/pointcloud_xyzrgb nodelet? Has anyone tested?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@doronhi, do you have any comments about this?

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

doronhi, 2020-12-14:
I didn't test which option is more stable or has better performance.

closed 2021-01-15 by RealSenseSupport