realsenseai/realsense-ros #723: D435 ros topic data difference between pointcloud

qiuchenli, 2019-04-10:
Dear Sir:
I'd like to know the difference between ros topics.
There are 2 topics contains pointcloud data ,which is /camera/depth_registerer/points and /camera/depth/color/points, I wonder the difference.
I find that the frame_id of 1st one is /camera/color/optical/frame, and the 2nd one is /camera/depth/optical/frame.
Any other suggestion?
Hope your reply！

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

@qiuchenli  Sorry for the late response. Did you try the latest ROS version to see if it's still issue for you? Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@qiuchenli Do you still need support on this? Thanks!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@qiuchenli Sorry that we didn't get response from you for weeks. Will close this ticket now. Please feel free to create new ticket if you still have any questions/issues. Thanks!

closed 2019-10-25 by RealSenseCustomerSupport