realsenseai/realsense-ros #725: error when try rosbag record it

jinfagang, 2019-04-11:
WIth librealsense 2.18 and ros wrapper, when rosbag record to record bags:

```
[ERROR] [1554971380.903484192]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.907463185]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.907785429]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.914533434]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.914900858]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.920987955]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.921352137]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.923855622]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.926548230]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.930398835]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.930732057]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.935613939]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.943120589]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.947236721]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.948965574]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: 8UC1).
[ERROR] [1554971380.953416284]: cv_bridge exception: '[8UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1554971380.970649630]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1554971380.987664006]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
*** Error in `/opt/ros/kinetic/lib/nodelet/nodelet': corrupted size vs. prev_size: 0x00007f2985ca4650 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f29c0cc47e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x83781)[0x7f29c0cd0781]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x179)[0x7f29c0cd1839]
/usr/lib/x86_64-linux-gnu/libogg.so.0(oggpackB_write+0x124)[0x7f29917592a4]
/usr/lib/x86_64-linux-gnu/libtheoraenc.so.1(+0x1acea)[0x7f2991b91cea]
/usr/lib/x86_64-linux-gnu/libtheoraenc.so.1(th_encode_ycbcr_in+0x3b8)[0x7f2991b94d98]
/opt/ros/kinetic/lib//libtheora_image_transport.so(+0xc1de)[0x7f2991dc21de]
/opt/ros/kinetic/lib//libtheora_image_transport.so(+0xf8f2)[0x7f2991dc58f2]
/opt/ros/kinetic/lib/libimage_transport.so(_ZNK15image_transport9Publisher7publishERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE+0x65)[0x7f29b03b2365]

```

martinakos, 2019-06-03:
I have the same problem. Have you found a solution?

mice85, 2019-06-05:
I have the same problem. I'm finding solutions.

jinfagang, 2019-06-05:
@martinakos @mice85  Turns out our deeplearning based monocular camera depth estimation algorithm out performs very much than realsense depth camera, we considering drop this sensor since it's depth really sucks....

jinfagang, 2019-06-05:
Meanwhile...  documentation really less and technique support is limited..... Too many simple questions can not get a practical answer...

doronhi, 2019-06-05:
@mice85 , @martinakos , Can you find your answer here: #315 ?

punchnami, 2021-01-22:
@jinfagang @doronhi 
To save the rosbag file, do we have to close the Rviz program (the one with point cloud and map cloud)  as well as its terminal first? and Is it normal for the rosbag file to be larger than 40 Gb (I was saving it while the Rviz program was opened and it continued saving to the point that the my_rosbagfile_1.bag.active was up to 40Gb) ?

I did terminate the program(Rviz) and tried to replay the my_rosbagfile_1.bag,nothing show in the new Rviz program.

MartyG-RealSense, 2021-01-23:
Hi @punchnami   I replied to your bag size question in your own case in the link below. 

https://github.com/IntelRealSense/realsense-ros/issues/1635#issuecomment-765445551

As this case is very old @punchnami  and a response has been provided to your question, i will close this case.  Thanks!

closed 2021-01-23 by MartyG-RealSense