realsenseai/realsense-ros #729: How to set the IMU frequency？

ltr-w, 2019-04-14:
When running    roslaunch realsense2_camera rs_camera.launch, I want to set the IMU parameter in the topic to accel_fps:200, gyro_fps:200, and the picture frame rate is set to color_fps:30. Which files do I need to modify?

doronhi, 2019-04-18:
I'm not sure I follow.
You should run: `roslaunch realsense2_camera rs_camera.launch accel_fps:=200 gyro_fps:=200`
Notice that 200Hz is not necessarily supported. Run `rs-enumerate-devices` to check which frame rates are available for each sensor.

beduffy, 2019-08-22:
When I use "linear_interpolation" with accel fps 250 and gyro fps 200, my /camera/imu topic is 400Hz and when I use "copy" it is 448Hz. Is this expected? 

I assumed it would be either 200 or 250 FPS but it's likely I just don't understand the imu_unite_method.

doronhi, 2019-08-28:
As explained in the README.md file, the "copy" method creates a united message for each incoming gyro or accel message. Therefor the output rate will be approximately 200+250 = 450 Hz.

For the "interpolation" method Consider the following scenario:
A1 means a message of type A (gyro or accel) at time 1
B2 means a message of type B (the other one then A) at time 2.
I drew the following sequence of messages. B's rate is higher (i.e. accel)
```
A messages: A1............A3................A6.......A8..........A10
B messages: B1......B2.......B4......B5......B7......B8......B9

time line:  t1......t2....t3.t4......t5.....t6.......t8......t9..t10...
```
Below is the content of the united messages at each time stamp:

t3: Arrived A3: send [interpolate(A3, A1 at t2), B2] with timestamp of t2.
t4: Arrived B4: send [A3, interpolate(B4, B2 at t3)] with timestamp of t3.
t5: Arrived B5: no new update for A so no message.
t6: Arrived A6: send [interpolate(A6, A3 at t5), B5] with timestamp of t5.

I tried to show that when 2 messages of the same type (B in this example) arrive with no message of the other type (A) in between, no new united message will be generated. Therefor, the rate of the united messages is expected to be about twice the lower frame rate (i.e. 2*200 = 400 Hz)

RealSenseCustomerSupport, 2019-11-12:
----------------------------------------------

Hi @ltr-w - doronhi has a pretty good explanation here.  Is there anything further you need?  I plan on closing out this thread.

closed 2019-11-19 by RealSenseCustomerSupport