realsenseai/realsense-ros #730: 2d box to 3d box offset

jinfagang, 2019-04-16:
Hi, I using object detection on RGB image, and try get the actullay 3D position of that object.

What I am doing is simply convert the 2d box center point's depth to get 3D coordinates.

then using a rectangle box to wrap it.

this is what I got;

[image]

As you can see, the point cloud is subscribed in realsense, but the position calculated were wrong.

what I want ask is:

1. What should I mapping a 2d bbox into 3D space in C++ properly, any snippets?
2. Another object can get correct position, this one offset, I think it caused by depth not smooth, the point I use exactly miss measured, **but why your realsense point cloud get it right?**

doronhi, 2019-04-18:
about question 2, I'm not sure, but could you try applying the hole_filling filter and see if you get improvement in your measurements? `roslaunch realsense2_camera rs_camera.launch filters:=hole_filling`

jinfagang, 2019-04-18:
@doronhi  Thanks for you reply. I saw that launch file I am using does not open any filters options, but your published out point cloud are accurate.

What I calcualte 3D position are accurate most of time, but also always missed distance (which is z), compare to your 3d point cloud published out, I simply took depth subscribed from launch file, then using the pixel as depth, to transform to 3D.

If you does not apply any operation about depth, then we should get same result ,but we were not now, what maybe the reason for this?

I can got almost accurate 3d box from 2d detection result on RGB image, but because of the poor quality of depth, I always miss target on 3D space, which is dangerous in our robot products. Except apply filtes in launch file, any more indications to fix this?

doronhi, 2019-04-18:
Maybe the best way for you is to compare your code to the way it's done in librealsense.
I would download the code and start following in the file [pointcloud.cpp](https://github.com/IntelRealSense/librealsense/blob/9c750f46c5623b912ea32a4998f3d3e81e9035d2/src/proc/pointcloud.cpp#L174)

jinfagang, 2019-04-18:
I think this the key clue: [/pointcloud.cpp#L34](https://github.com/IntelRealSense/librealsense/blob/9c750f46c5623b912ea32a4998f3d3e81e9035d2/src/proc/pointcloud.cpp#L34)

where did this function locate?

doronhi, 2019-04-18:
[here](https://github.com/IntelRealSense/librealsense/blob/9c750f46c5623b912ea32a4998f3d3e81e9035d2/include/librealsense2/rsutil.h#L68)

jinfagang, 2019-04-19:
@doronhi  Many thanks for your reply. It seems that my conversion was slightly different than librealsense. I haven't consider distortion (simply using fx_, fy_, cx_, cy_). Then there is an question:

1. when should we consider the distortion? 
2. Can we simply using this function in librealsense to deproject the point?

jinfagang, 2019-04-19:
BTW, I can not get a rs_intrinsics object in ros, since I can not open a stream on realsense if ros launch using it.

I can only subscriber camera_info topic to get the intrincs, how to convert it into rs_intrinsics object?

jinfagang, 2019-04-19:
OK, I logged out camera info, the distortion model is: plumb_bob, which is not either of those:

- RS2_DISTORTION_INVERSE_BROWN_CONRADY
- RS2_DISTORTION_KANNALA_BRANDT4
- RS2_DISTORTION_FTHETA

what should I do now?

jinfagang, 2019-04-19:
Full camera info is:

```
header: 
  seq: 5969
  stamp: 
    secs: 1555496142
    nsecs: 880019678
  frame_id: "camera_color_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [618.5665893554688, 0.0, 329.1777038574219, 0.0, 618.917724609375, 249.52749633789062, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [618.5665893554688, 0.0, 329.1777038574219, 0.0, 0.0, 618.917724609375, 249.52749633789062, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---

```

Domihk, 2019-05-02:
@jinfagang any update on this issue? I'm also interested in editing/processing the RPG before alignment and coloring of the pointcloud is made. But I don't see clearly the proper place to do this.

Neither I think we can subscribe to the color and depth separately cause we will receive un-aligned data.

jinfagang, 2019-05-02:
@Domihk  You have to align depth and image, otherwise you can not get accurate depth of pixel you interested in RGB color image.

My problem is, the depth image subscribed from sensor does not quite good and the 3d point calculated out were mismeasured then librealsense calculated point cloud.

I just can not got same result as their calculated point cloud

Domihk, 2019-05-03:
@jinfagang So you mean the processing of the color image should be done after alignment instead of before? What could be the reason for this?

Actually I cannot find how to align the images in their ROS wrapper. (in their c++ sdk is straight forward)

jinfagang, 2019-05-03:
@Domihk You can compare the depth and RGB image (stack togather), they can not fully coinside.

Domihk, 2019-05-03:
@jinfagang I know already that they are not aligned.

I need to do image processing on the color image **before** alignment, and after I do this, I need to call the alignment function. But I don't know how to call the alignment function in the ROS wrapper. Do you know how?

jinfagang, 2019-05-04:
@Domihk  Align has nothing to do with your process with RGB image. As long as you does not reshape your original RGB image, the pixel coordinates still same. For calling align, there were no way to call it, but you can set in ros launch, then you can subscribe topic which is aligned already

closed 2024-03-29 by MartyG-RealSense