realsenseai/realsense-ros #733: Roslaunch rs_t265.launch error

KentaKawamata, 2019-04-17:
Dear Sir:
I'd like to know the solution.
Errors are keep popping up using T265.

## System setup

|||
|:--:|:--:|
|Ubuntu|16.04.5|
|kernel|4.15.0-47-generic|
|ROS| ros-kinetic|
|T265 firmware|0.0.18.5448|
|librealsense|2.20.0|
|ros wrapper|2.2.3|

## Sumary

```
$ roslaunch realsense2_camera rs_t265.launch serial_no:=<serial number>  name:=T265
```

```
... logging to /root/.ros/log/021a0f6c-6103-11e9-b12a-0242ac110002/roslaunch-86d96596aeea-3115.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://86d96596aeea:36783/

SUMMARY
========

PARAMETERS
 * /T265/realsense2_camera/accel_fps: 62
 * /T265/realsense2_camera/accel_frame_id: T265_accel_frame
 * /T265/realsense2_camera/accel_optical_frame_id: T265_accel_optica...
 * /T265/realsense2_camera/align_depth: False
 * /T265/realsense2_camera/aligned_depth_to_color_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_fisheye1_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_fisheye2_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_fisheye_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_infra1_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_infra2_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/base_frame_id: T265_link
 * /T265/realsense2_camera/calib_odom_file: 
 * /T265/realsense2_camera/clip_distance: -2.0
 * /T265/realsense2_camera/color_fps: 30
 * /T265/realsense2_camera/color_frame_id: T265_color_frame
 * /T265/realsense2_camera/color_height: 480
 * /T265/realsense2_camera/color_optical_frame_id: T265_color_optica...
 * /T265/realsense2_camera/color_width: 640
 * /T265/realsense2_camera/depth_fps: 30
 * /T265/realsense2_camera/depth_frame_id: T265_depth_frame
 * /T265/realsense2_camera/depth_height: 480
 * /T265/realsense2_camera/depth_optical_frame_id: T265_depth_optica...
 * /T265/realsense2_camera/depth_width: 640
 * /T265/realsense2_camera/enable_accel: True
 * /T265/realsense2_camera/enable_color: True
 * /T265/realsense2_camera/enable_depth: True
 * /T265/realsense2_camera/enable_fisheye1: True
 * /T265/realsense2_camera/enable_fisheye2: True
 * /T265/realsense2_camera/enable_fisheye: True
 * /T265/realsense2_camera/enable_gyro: True
 * /T265/realsense2_camera/enable_infra1: True
 * /T265/realsense2_camera/enable_infra2: True
 * /T265/realsense2_camera/enable_pointcloud: False
 * /T265/realsense2_camera/enable_sync: False
 * /T265/realsense2_camera/filters: 
 * /T265/realsense2_camera/fisheye1_frame_id: T265_fisheye1_frame
 * /T265/realsense2_camera/fisheye1_optical_frame_id: T265_fisheye1_opt...
 * /T265/realsense2_camera/fisheye2_frame_id: T265_fisheye2_frame
 * /T265/realsense2_camera/fisheye2_optical_frame_id: T265_fisheye2_opt...
 * /T265/realsense2_camera/fisheye_fps: 30
 * /T265/realsense2_camera/fisheye_frame_id: T265_fisheye_frame
 * /T265/realsense2_camera/fisheye_height: 800
 * /T265/realsense2_camera/fisheye_optical_frame_id: T265_fisheye_opti...
 * /T265/realsense2_camera/fisheye_width: 848
 * /T265/realsense2_camera/gyro_fps: 200
 * /T265/realsense2_camera/gyro_frame_id: T265_gyro_frame
 * /T265/realsense2_camera/gyro_optical_frame_id: T265_gyro_optical...
 * /T265/realsense2_camera/infra1_frame_id: T265_infra1_frame
 * /T265/realsense2_camera/infra1_optical_frame_id: T265_infra1_optic...
 * /T265/realsense2_camera/infra2_frame_id: T265_infra2_frame
 * /T265/realsense2_camera/infra2_optical_frame_id: T265_infra2_optic...
 * /T265/realsense2_camera/infra_fps: 30
 * /T265/realsense2_camera/infra_height: 480
 * /T265/realsense2_camera/infra_width: 640
 * /T265/realsense2_camera/initial_reset: False
 * /T265/realsense2_camera/json_file_path: 
 * /T265/realsense2_camera/linear_accel_cov: 0.01
 * /T265/realsense2_camera/odom_frame_id: T265_odom_frame
 * /T265/realsense2_camera/pointcloud_texture_index: 0
 * /T265/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /T265/realsense2_camera/pose_frame_id: T265_pose_frame
 * /T265/realsense2_camera/pose_optical_frame_id: T265_pose_optical...
 * /T265/realsense2_camera/publish_odom_tf: True
 * /T265/realsense2_camera/rosbag_filename: 
 * /T265/realsense2_camera/serial_no: 845412110931
 * /T265/realsense2_camera/topic_odom_in: odom_in
 * /T265/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /T265/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3125]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 021a0f6c-6103-11e9-b12a-0242ac110002
process[rosout-1]: started with pid [3138]
started core service [/rosout]
process[T265/realsense2_camera_manager-2]: started with pid [3155]
[ INFO] [1555500117.363303216]: Initializing nodelet with 8 worker threads.
process[T265/realsense2_camera-3]: started with pid [3156]
[ INFO] [1555500117.637760516]: RealSense ROS v2.2.3
[ INFO] [1555500117.637824889]: Running with LibRealSense v2.19.2
[ INFO] [1555500117.889624976]: Checking new devices...
 17/04 11:21:58,226 WARNING [140645551679360] (backend-hid.cpp:610) open() failed!
 17/04 11:21:58,278 WARNING [140644778997504] (types.cpp:59) hwmon command 0x4f failed. Error type: No data to return (-21).
 17/04 11:21:58,283 WARNING [140645551679360] (types.cpp:59) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1555500118.298764615]: No calib_odom_file. No input odometry accepted.
[ INFO] [1555500118.298883129]: getParameters...
[ INFO] [1555500118.299747458]: No calib_odom_file. No input odometry accepted.
[ INFO] [1555500118.301617576]: getParameters...
[ERROR] [1555500118.302523396]: An exception has been thrown: Character [?] is not valid as the first character in Graph Resource Name [?U?
[FATAL] [1555500118.317407768]: Failed to load nodelet '/T265/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
 17/04 11:21:58,317 ERROR [140644778997504] (context.cpp:432) Exception thrown from user callback handler
[T265/realsense2_camera-3] process has died [pid 3156, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/root/.ros/log/021a0f6c-6103-11e9-b12a-0242ac110002/T265-realsense2_camera-3.log].
log file: /root/.ros/log/021a0f6c-6103-11e9-b12a-0242ac110002/T265-realsense2_camera-3*.log
```

GNREX, 2019-04-24:
Needs resolving. Camera works once out of 10 tries. Probably a race condition.

joekeo, 2019-04-24:
I am having the same problem. I do have the magic line in my launch file:
<arg name="initial_reset"       default="true"/>
<arg name="initial_reset"            value="$(arg initial_reset)"/>

Still it is very unreliable to be initialised

schmidtp1, 2019-04-30:
Thank you for reporting this issue and thank you for your patience.
We are currently debugging it.
The node still works with librealsense 2.16.1 but this would mean using an old FW missing several improvements since then.

schmidtp1, 2019-05-01:
This fixes the issue for me: https://github.com/intel-ros/realsense/pull/752

joekeo, 2019-05-03:
Many thanks for this update!

KentaKawamata, 2019-05-04:
Thanks for this update.
I confirmed that T265 works with the wrapper.

closed 2019-05-04 by KentaKawamata